open62541/open62541 #5687: High frequent "Bad no subscription" errors in client

tboeckel, 2023-03-16:
## Description
From time to time we are facing the problem that an open62541 based client connects to an open62541 based server and then obviously something goes wrong with the subscription of variables. We set up a subscription and the put an arbitrary number of monitored items into that subscription. Usually this all works perfectly, but something something goes wrong (without any readable notice) and then we get tons of these "BadNoSubscription" log messages right after a complain about a dropped StatusChangeNotification:

[11:03:51.731] WARN [warning|client   ] Dropped a StatusChangeNotification since no callback is registered
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.731] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.732] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
[11:03:51.732] WARN [warning|client   ] Received Publish Response with code BadNoSubscription
.....

Upon the next start of our application everything is working again, at least in 98 of 100 times. No changes on either side inbetween. But in this situation neither side is really usable.

What is causing this behaviour and what can we do about it?

Used CMake options:
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OPENSSL
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=ON
UA_MULTITHREADING:STRING=100

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.3.5
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 20.04, Ubuntu 22.04, custom Yocto Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2023-04-25:
This issue is becoming a real show stopper. Sometimes the applications start properly on each try, sometimes I get thhis issue each and every time the applications start. I really get the impression the issue is dependend on aliens, moon phases or what ever.
It would be really nice if someone could please take a look at this.

tboeckel, 2023-06-05:
Although the issue seemed to have vanished for some time it occures more often again now. Even for client applications which do NOT subscribe ANY variables from the remote servers. How can this happen?

We really need to get this issue resolved. Please help me!!

tboeckel, 2023-06-05:
I must correct myself. The effect still only happens for client application which actually do subscribe values from the remote server. I was mislead by a misinterpreted debug output. Sorry for that.
But the basic question remains: what is causing this error? Why is the subscription transferred to another session? At least the is the location where the status change notification is thrown.

tboeckel, 2023-09-01:
In the meantime I added a StatusChange notification callback. The only thing it is able to output is a statuscode of UA_STATUSCODE_GOODSUBSCRIPTIONTRANSFERRED. Why does this happen? And what can I do about that or how should this situation be handled?

VictorManKBO, 2023-09-04:
I have this problem too. When clients connect at different times, everything works. But, if you turn off and turn on the server, then the clients immediately restore communication and somewhere such an error (internal) error occurs. With all this, retval = GOOD.
When only one client works - no such problem is noticed. My version **v1.3.4**

```
4/9 13:0:45.874  OpcUaClient creted!
4/9 13:0:45.927  CheckConnection() : BadDisconnect
4/9 13:0:45.927  [warn/userland]	AcceptAll Certificate Verification. Any remote certificate will be accepted.
4/9 13:0:45.932  [info/channel]	Connection 816 | SecureChannel 3 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
4/9 13:0:45.932  [info/client]	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
4/9 13:0:45.933  [info/client]	Selected endpoint 0 in URL opc.tcp://192.168.220.43:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
4/9 13:0:45.934  [info/client]	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
4/9 13:0:45.935  [info/client]	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
4/9 13:0:45.936  [info/client]	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
4/9 13:0:45.937  Reconnect() : connected to 192.168.220.43:4840
4/9 13:0:47.953  subscription #3 created  with status Good
4/9 13:0:56.775  [warn/channel]	Connection 816 | SecureChannel 3 | Receiving the response failed with StatusCode BadConnectionClosed
4/9 13:0:56.776  [warn/client]	Received Publish Response with code BadSecureChannelClosed
4/9 13:0:56.777  [warn/client]	Received Publish Response with code BadSecureChannelClosed
4/9 13:0:56.778  [warn/client]	Received Publish Response with code BadSecureChannelClosed
4/9 13:0:56.784  [warn/channel]	Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
4/9 13:0:56.784  [info/client]	Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
4/9 13:0:56.785  CheckConnection() : Good
4/9 13:0:56.836  CheckConnection() : Good
4/9 13:0:56.836  [info/client]	SecureChannel must be connected before sending requests
4/9 13:0:56.837  [info/client]	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
4/9 13:0:56.837  [warn/userland]	AcceptAll Certificate Verification. Any remote certificate will be accepted.
4/9 13:1:1.838  [warn/network]	Connection to opc.tcp://192.168.220.43:4840 timed out
4/9 13:1:1.839  [info/client]	Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadTimeout
4/9 13:1:1.839  Reconnect() : connection error with 192.168.220.43:4840(BadTimeout)
4/9 13:1:1.840  CheckConnection() : BadTimeout
...........................	BadTimeout
4/9 13:1:27.13  CheckConnection() : BadTimeout
4/9 13:1:27.13  [info/client]	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadTimeout
4/9 13:1:27.14  [warn/userland]	AcceptAll Certificate Verification. Any remote certificate will be accepted.
4/9 13:1:27.19  [info/channel]	Connection 816 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
4/9 13:1:27.20  [info/client]	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
4/9 13:1:27.21  [info/client]	Selected endpoint 0 in URL opc.tcp://192.168.220.43:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
4/9 13:1:27.22  [info/client]	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
4/9 13:1:27.24  [info/client]	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
4/9 13:1:27.25  [info/client]	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
4/9 13:1:27.26  Reconnect() : connected to 192.168.220.43:4840
4/9 13:1:29.29  subscription #1 creted with status Good
4/9 13:1:29.785  [warn/client]	Dropped a StatusChangeNotification since no callback is registered
4/9 13:1:29.786  [info/client]	Received a ServiceFault response
4/9 13:1:29.787  [info/client]	The ServiceResult has the StatusCode BadNoSubscription
4/9 13:1:29.787  [warn/client]	Received Publish Response with code BadNoSubscription

```

tboeckel, 2023-09-08:
Unfortunately this patch does not solve the issue for me. I still get these high frequent "BadNoSubscription" messages.

tboeckel, 2023-09-11:
In the meantime I found a workaround for this issue. It is not perfect and I really like to have a better approach, but for the time being I can live with this "fix".
If the status change notification tells about a successfully transferred subscription I simply abort the connection to the server and restart the connection from scratch. Then, on the second try, everything works out perfectly and the subscription is no longer transferred and hence no high frequent BadNoSubscription messages occur.

But why is the transfer of a subscription from one session to another session (which has NOT been initiated by my application) is required or necessary at all? My client sets up a single connection to a server. This connection should result in one single session. From my point of view there is no space for a transfer to another session, at least not to a session that my application never caused to be set up and hence cannot control.

oliver, 2024-02-08:
I have this problem as well, and it appears to be caused by having UaExpert running in parallel. Apparently UaExpert will after reconnection try to transfer its subscription from the previous session; but it will use the SubscriptionId that it used in the previous session, even though that SubscriptionId might now be used by some other client!

This is quite similar to the description by @VictorManKBO in https://github.com/open62541/open62541/issues/5687#issuecomment-1706682794 .

I am able to reproduce this problem reliably, like this:
- start my server
- start UaExpert
  - it will now get SubscriptionId 1
- stop my server
- start my server and immediately start my client
  - my client will now  get SubscriptionId 1 (because it is now the first client to connect)
- after ca. one second, UaExpert will automatically reconnect as well
- now UaExpert will send a TransferSubscriptionsRequest message, for SubscriptionId 1 !
- and my server will happily transfer the subscription (that now belongs to my client) to UaExpert
  - my client will receive a StatusChangeNotification message (which causes the "Dropped a StatusChangeNotification since no callback is registered" log output); and I guess the "Received Publish Response with code BadNoSubscription" logs are followup-errors.

In this case, server and client are using open62541 1.3.9 and open62541pp 0.11.0, under Debian 12 (Bookworm). UaExpert is in version 1.7.1 540.

I am not experienced with OPC-UA, so cannot really say who is at fault here:
- it seems suspicious that UaExpert simply assumes that its previous SubscriptionId is still valid. But maybe there is something in the specification that allows this?
- the spec states: "The Server shall validate that the Client of that Session is operating on behalf of the same user" (https://reference.opcfoundation.org/Core/Part4/v104/docs/5.13.7).
- the open62541 server code apparently performs exactly this user check: "For TransferSubscriptions, we check whether the transfer happens between Sessions for the same user." (https://github.com/open62541/open62541/blob/v1.3.9/plugins/ua_accesscontrol_default.c#L17)
- I use anonymous authentication in my client and in UaExpert, so I guess the user is assumed to be the same?

Anyway, I guess the solution is not use UaExpert, or maybe set up a separate user account for UaExpert, or do not use the automatic reconnection in UaExpert?

tboeckel, 2024-10-21:
I recently reported this issue to Unified Automation, and this is the answer I received from them:

-----------------------------------------------------------------------
TransferSubscription is a normal mechanism in OPC UA to recover from connection loss without data loss, even during longer network interruptions.

We have never had this problem with servers built with our server SDKs. There are several mechanisms in OPC UA that prevent the problem described. It seems to be based on a collection of security bugs in the open62541 server implementation. Just by reading the linked topic for 5 minutes I found two obvious security flaws.

Writing a ticket for UaExpert is like accusing an FTP client of being able to read all the files on a system if the user is logged in as root without security.

Best Regards,
Support Team
Unified Automation GmbH
-----------------------------------------------------------------------

What security bugs are they talking about?

tboeckel, 2024-10-24:
I got a further response from Unified Automation regarding my question about the security bugs:

-------------------------------------------------------------------------
für das das offensichlichste Problem haben wir uns gewundert dass das nicht durch das Compliance Test Tool der OPC Foundation gefunden wird. Das haben wir jetzt mit der Compliance Arbeitsgruppe geklärt und da hat in der Tat ein Testcase gefehlt. Der wird jetzt ergänzt. Dieses Problem ist dass der OPC UA Server das TransferSubscription für einen Anonymous User akzeptiert. Das war immer verboten. Vor kurzem wurde eine Ausnahme eingeführt bei der Anonymous erlaubt ist wenn der Client mit einem mindestens signierten Kanal kommunizert und dann geprüft werden kann ob der Client der neuen Session identisch ist mit dem Client der die vorherige Session angelegt hatte. Aber ein TransferSubscription mit Anonymous User über einen Channel ohne Security (NONE) ist definiitiv nicht eraubt.

Ansonsten ist es nicht unsere Aufgabe die Fehler in Bibliotheken zu analysieren die nicht von uns sind. Da müssen die zuständigen Entwickler schon selbst die Spezifikation lesen.

Best Regards,
Support Team
Unified Automation GmbH
-------------------------------------------------------------------------

I hope this helps to fix this issue.