open62541/open62541 #5689: Compilation for Espressif ESP32 FREERTOSLWIP esp-idf 5.0 fails

klaus-liebler, 2023-03-16:
## Description

## Background Information / Reproduction Steps
I followed the recommended steps described in https://www.open62541.org/doc/master/building.html#building-for-specific-architectures to create the amalgamated file open62541.c/.h. My settings in cmake-gui are:
[image]

When I tried to compile a simple server example I get bazillions of errors, mainly targeting undeclared functions UA_<some function>_POSIX, e.g.:

```
C:/repos/espidf-opcua/components/open62541lib/open62541.c:83624:27: error: implicit declaration of function 'UA_EventLoop_new_POSIX' [-Werror=implicit-function-declaration]
83624 |         conf->eventLoop = UA_EventLoop_new_POSIX(&conf->logger);
```

Well, the declaration for these functions is not available on other system than POSIX and WIN32, especially not on FREERTOSLWIP (see file eventloop.h):
```c
/**
 * POSIX-Specific Implementation
 * -----------------------------
 * The POSIX compatibility of WIN32 is 'close enough'. So a joint implementation
 * is provided. */

#if defined(UA_ARCHITECTURE_POSIX) || defined(UA_ARCHITECTURE_WIN32)

UA_EXPORT UA_EventLoop *
UA_EventLoop_new_POSIX(const UA_Logger *logger);
//other functions to follow...
```

How can this problem be solved?

Thank you for your help!

Regards, Klaus

Used CMake options:

```bash
Commandline options:
-DCLANG_FORMAT_EXE:FILEPATH="CLANG_FORMAT_EXE-NOTFOUND" 

Cache file:
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.5
 - [x] Other OPC UA SDKs used (client or server): server
 - [x] Operating system: Build Host: Windows 11, Target: Espressif ESP32 / FreeRTOS+Lwip / esp-idf 5.0
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

KnowingBat, 2023-03-17:
Can you try with the version 1.3? I had the same problem, but after a quick change to 1.3, almost everything was fine. 
I'm not a contributor so maybe i could be wrong... but anyway give it a try.

klaus-liebler, 2023-03-17:
Tried with success! Thank you so much, KnowingBat!
The only manual change was in open62541.h file: "
```c
#include <task.h>" --> "#include <freertos/task.h>
```
Furthermore, as the binary got a little bit larger, I had to select "Single Factory all (large), no OTA" in menuconfig->partition table.

I would be great if the master branch can be compiled for ESP32 in the future!

Thank you for this great project!

Regards, Klaus

KnowingBat, 2023-03-17:
Great news, Klaus!
As far as i know, the contributors are cleaning the the arch support interfaces on master. 
i think the 1.3 branch will be the only reference for freeRTOS/embedded world.

KnowingBat, 2023-03-22:
Hello @klaus-liebler.
Have you experienced RAM problems when running a server on your ESP32?
I want to ask because I cannot get my server to run because of RAM usage (i guess). 
The function
```C
UA_Server *server = UA_Server_new();
``` 
is initializing the server to ```NULL```. I dig into the functions and find out that ```namespace0_generated(server)``` is returning the value ```UA_BADOUTOFMEMORY```.
I also tried to set the namespace0 to MINIMAL and to increase the heap up to 470kB. Still does not work

klaus-liebler, 2023-03-22:
Hi,

yes, I had these issues as well. Luckily, I am working with an esp32 s3 n8r2 with 2MB of spi ram. After having activated it, everything was running smoothly... 

Regards, Klaus