open62541/open62541 #5690: After changing from version 1.2 to 1.35, certificate validation failed. What is the reason for this?

liutt1312, 2023-03-17:
## Description

## Background Information / Reproduction Steps

After changing from version 1.2 to 1.35, certificate validation failed. What is the reason for this?
[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2023-03-17:
Try setting up the certificate trust list in the config. Since v1.3.4 an empty trust list will result in certificate rejection (as it should).

liutt1312, 2023-03-17:
> Try setting up the certificate trust list in the config. Since v1.3.4 an empty trust list will result in certificate rejection (as it should).

The certificate used is in the server trust list

liutt1312, 2023-03-17:
> > Try setting up the certificate trust list in the config. Since v1.3.4 an empty trust list will result in certificate rejection (as it should).
> 
> The certificate used is in the server trust list

The only difference between the same code and certificate is the library version

neojaw, 2023-03-17:
> The certificate used is in the server trust list

And is the server certificate in the client trust list?

liutt1312, 2023-03-17:
> > The certificate used is in the server trust list
> 
> And is the server certificate in the client trust list?

Do you need this step? It is also possible to add a server certificate when using version 1.2

neojaw, 2023-03-17:
Yes, this is needed. Up to [v1.3.4](https://github.com/open62541/open62541/releases/tag/v1.3.4) an empty trust list caused the client to just accept all server certificates but then this was changed/fixed.

liutt1312, 2023-03-17:
> Yes, this is needed. Up to [v1.3.4](https://github.com/open62541/open62541/releases/tag/v1.3.4) an empty trust list caused the client to just accept all server certificates but then this was changed/fixed.

okay, thank you

closed 2023-03-17 by liutt1312

liutt1312, 2023-03-17:
> Yes, this is needed. Up to [v1.3.4](https://github.com/open62541/open62541/releases/tag/v1.3.4) an empty trust list caused the client to just accept all server certificates but then this was changed/fixed.

Can this problem be solved？
After cross compilation, this error occurs when running on arm linux. What is the reason for this error #5691