open62541/open62541 #5701: Single Callback for whole subscription possible? / processDataChangeNotification is slow for subscriptions with many monitored items

JulZimmermann, 2023-03-20:
Hi,

this is a follow-up on #5053 and #4318.

The current implementation of `processDataChangeNotification` in `ua_client_subscriptions.c` is not really performant when subscribing to multiple (e.g. 10000) monitored items.

#4318 already asks about the possibility to get notified via a single callback call for a subscription, instead of callbacks for every monitored item. Unfortunately, this question was never answered.

#5053 and #5068 changed the implementation of `processDataChangeNotification` to be more performant by using a hash map. However, these changes are not in the latest release and only exist in the main branch.

I have a couple of questions now:

1. Is there some (hidden) way I didn't see, to get notified only once per subscription instead of for every monitored item? This would allow bypassing `processDataChangeNotification` altogether. In my (and probably in many other) use cases, I use a hashmap for managing the monitored items anyway, already at application level. The logic in `processDataChangeNotification` (even with a more performant hash map implementation) is therefore not needed.
2. If this is currently not possible: Would it be possible to add this functionality?
3. Could #5068 be backported to the 1.3 branch?
4. Is it "save" to use the master branch?

Thanks!

JulZimmermann, 2023-11-30:
Any news on this?

o6-nbeier, 2026-10-20:
Thank you for opening this issue and for the details you provided.

It describes the same problem as the earlier issue #4318, so we are closing this one as a duplicate to keep the discussion in one place.

closed 2026-10-20 by o6-nbeier