open62541/open62541 #5702: Erroneous client behavior when connection is lost during server browsing

AlexOrkin, 2023-03-20:
## Description
Hello.
I use the open62541 library v.1.3.3 in Linux, with Qt5.12 / GCC 6.3.0. My code browses an OPC server in the another network node. When the connection gets lost during synchronous browsing, there is no error and client status is good, while the data received is not complete; I can't distinguish this situation from normal work completion. Could you please explain if I'm doing something wrong. Thank you.
## Code
```
#include "open62541.h"
#include <QThread>
#include <QTime>

static char serverUrl[] = "opc.tcp://192.168.31.101:4841";
static UA_Client *client;
static UA_StatusCode nodeIter (UA_NodeId childId, UA_Boolean isInverse, UA_NodeId /*referenceTypeId*/, void *handle)
{
    static unsigned seqNo = 0;
    if(isInverse)
        return UA_STATUSCODE_GOOD;
    if (childId.namespaceIndex != 0)
    {
        printf(" ---%05u %s) NodeId %u, %u\n", ++seqNo,
               QTime::currentTime().toString("hh:mm:ss.zzz").toStdString().c_str(),
               childId.namespaceIndex, childId.identifier.numeric);
        fflush(stdout);
        UA_Client_forEachChildNodeCall(client, childId, nodeIter, handle);
        if(seqNo==9090) QThread::sleep(10); // pause to break connection manually
    }
    return UA_STATUSCODE_GOOD;
}

int main(int argc, char *argv[])
{
    client = UA_Client_new();
    UA_ClientConfig *config = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(config);
    printf ("+++++Connecting...\n"); fflush(stdout);
    UA_StatusCode retval = UA_Client_connect(client, serverUrl);
    if (!UA_StatusCode_isGood(retval))
        { UA_Client_delete(client); return -1; }
    printf ("+++++Browsing server...\n"); fflush(stdout);
        // browse server
    UA_Client_forEachChildNodeCall(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), nodeIter, NULL);
    UA_StatusCode status;
    UA_Client_getState(client, NULL, NULL, &status);
    printf ("+++++Done: status = %X (%s)\n", status, UA_StatusCode_name(status)); fflush(stdout);
    UA_Client_delete(client);
    return 0;
}
```
## Log when the connection is not lost
```
[2023-03-20 19:39:16.450 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
+++++Connecting...
[2023-03-20 19:39:16.450 (UTC+0200)] info/eventloop	Starting the EventLoop
[2023-03-20 19:39:16.450 (UTC+0200)] info/network	TCP 4	| New connection to "192.168.31.101" on port 4841
[2023-03-20 19:39:16.451 (UTC+0200)] info/channel	TCP 4	| SC 161	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-03-20 19:39:16.451 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-03-20 19:39:16.451 (UTC+0200)] info/client	Selected endpoint 0 in URL opc.tcp://192.168.31.101:4841 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-03-20 19:39:16.451 (UTC+0200)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-03-20 19:39:16.452 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-03-20 19:39:16.453 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
+++++Browsing server...
 ---00001 19:39:16.453) NodeId 1, 1
 ---00002 19:39:16.454) NodeId 1, 2
 ---00003 19:39:16.454) NodeId 1, 4
  ......
 ---09089 19:39:21.686) NodeId 1, 9094
 ---09090 19:39:21.686) NodeId 1, 9083
 ---09091 19:39:31.687) NodeId 1, 9100
 ---09092 19:39:31.687) NodeId 1, 9089
 ---09093 19:39:31.687) NodeId 1, 9078
 ---09094 19:39:31.688) NodeId 1, 9095
 ---09095 19:39:31.688) NodeId 1, 9084
 ---09096 19:39:31.688) NodeId 1, 9101
 ---09097 19:39:31.689) NodeId 1, 9090
 ---09098 19:39:31.689) NodeId 1, 9079
 ---09099 19:39:31.689) NodeId 1, 9102
 ---09100 19:39:31.690) NodeId 1, 9103
 ---09101 19:39:31.690) NodeId 1, 9104
 ---09102 19:39:31.690) NodeId 1, 9105
 ---09103 19:39:31.691) NodeId 1, 9162
  ......
 ---13518 19:39:34.586) NodeId 1, 13519
 ---13519 19:39:34.586) NodeId 1, 13508
+++++Done: status = 0 (Good)
[2023-03-20 19:39:34.586 (UTC+0200)] info/channel	TCP 0	| SC 0	| Closed the SecureChannel
[2023-03-20 19:39:34.586 (UTC+0200)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2023-03-20 19:39:34.586 (UTC+0200)] info/network	TCP 4	| Socket closed
[2023-03-20 19:39:34.685 (UTC+0200)] info/eventloop	Stopping the EventLoop
[2023-03-20 19:39:34.685 (UTC+0200)] info/network	UDP	| Shutting down the ConnectionManager
[2023-03-20 19:39:34.685 (UTC+0200)] info/network	TCP	| Shutting down the ConnectionManager
[2023-03-20 19:39:34.686 (UTC+0200)] info/eventloop	The EventLoop has stopped
```
## Log when the connection is lost when browsing
```
[2023-03-20 19:35:40.115 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
+++++Connecting...
[2023-03-20 19:35:40.115 (UTC+0200)] info/eventloop	Starting the EventLoop
[2023-03-20 19:35:40.116 (UTC+0200)] info/network	TCP 4	| New connection to "192.168.31.101" on port 4841
[2023-03-20 19:35:40.117 (UTC+0200)] info/channel	TCP 4	| SC 160	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-03-20 19:35:40.118 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-03-20 19:35:40.128 (UTC+0200)] info/client	Selected endpoint 0 in URL opc.tcp://192.168.31.101:4841 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-03-20 19:35:40.128 (UTC+0200)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-03-20 19:35:40.129 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-03-20 19:35:40.129 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
+++++Browsing server...
 ---00001 19:35:40.129) NodeId 1, 1
 ---00002 19:35:40.130) NodeId 1, 2
 ---00003 19:35:40.130) NodeId 1, 4
   .....
 ---09089 19:35:44.955) NodeId 1, 9094
 ---09090 19:35:44.955) NodeId 1, 9083
      --------- connection lost here -----------
 ---09091 19:35:54.958) NodeId 1, 9100
 ---09092 19:35:59.960) NodeId 1, 9089
 ---09093 19:36:04.960) NodeId 1, 9078
 ---09094 19:36:09.977) NodeId 1, 9095
 ---09095 19:36:14.977) NodeId 1, 9084
 ---09096 19:36:19.985) NodeId 1, 9101
 ---09097 19:36:24.987) NodeId 1, 9090
 ---09098 19:36:30.001) NodeId 1, 9079
 ---09099 19:36:35.014) NodeId 1, 9102
 ---09100 19:36:40.014) NodeId 1, 9103
 ---09101 19:36:45.015) NodeId 1, 9104
 ---09102 19:36:50.017) NodeId 1, 13498
+++++Done: status = 0 (Good)
[2023-03-20 19:37:00.031 (UTC+0200)] info/channel	TCP 0	| SC 0	| Closed the SecureChannel
[2023-03-20 19:37:00.031 (UTC+0200)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2023-03-20 19:37:00.031 (UTC+0200)] info/network	TCP 4	| Socket closed
[2023-03-20 19:37:00.142 (UTC+0200)] info/eventloop	Stopping the EventLoop
[2023-03-20 19:37:00.142 (UTC+0200)] info/network	UDP	| Shutting down the ConnectionManager
[2023-03-20 19:37:00.142 (UTC+0200)] info/network	TCP	| Shutting down the ConnectionManager
[2023-03-20 19:37:00.142 (UTC+0200)] info/eventloop	The EventLoop has stopped
```

jpfr, 2023-04-06:
It is possible that the qt wrapper uses browse/browseNext.
So that browsing is split into several responses.
Then the partial results will still be "good".

In your logs, do you close the SecureChannel or is this done automatically?

o6-nbeier, 2026-10-20:
Closing for no activity.

closed 2026-10-20 by o6-nbeier