open62541/open62541 #5707: Failing unit test check_pubsub_encoding_custom

keba-estr, 2023-03-22:
## Description

Since PR #5681 the new unit test check_pubsub_encoding_custom fails (sporadically?) with some heap-buffer-overflow message.

Example build: https://github.com/open62541/open62541/actions/runs/4489318805/jobs/7894990967?pr=5697

@basyskom-dege fyi

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-03-23:
I ran into similar issues and fixed them as part of #5693.
Please verify if this is resolved and close.

keba-estr, 2023-03-23:
Problem does not occur any more after update to new commits.

closed 2023-03-23 by keba-estr