open62541/open62541 #5711: Unable to run application which utilizes Asset Management Basics (AMB) node-set

DrMatthiasArnold, 2023-03-27:
## Description
When trying to load the nodes defined in the AMB companion specification node-set file (after successfully loading nodes from UA core and DI) the application crashes, because it is unable to find the parent nodes for several child nodes (they are obviuosly not created yet).
This applies to the children of Objects/Aliases as well as children of Objects/Locations (like HierarchicalLocations or OperationalLocations).
In all observed cases the parent-child relationship is defined by an Organizes reference instead of HasComponent or other Aggregates references.

## Background Information / Reproduction Steps
Build a simple applciation which includes DI and AMB.
The applcation crashes during startup.
When the AMB node-set is removed or all child nodes below Aliases or Locations are removed from the AMB node-set file, the application runs.

## Checklist
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.X
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

GoetzGoerisch, 2023-03-27:
Could you please point out a few of these node, which does not load?

DrMatthiasArnold, 2023-03-27:
The nodes under Objects/Aliases & Objects/Locations which are provided by AMB and referenced via Organizes as depitced in the images below:
[image]
[image]
[image]

AndreasSchorpp, 2023-03-28:
The problem caused because of an older version of the Opc.Ua.Nodeset2.xml file in the deps/ua-nodeset/Schema folder. After updating the ua-nodeset folder from git the AMB can be loaded into the server.