open62541/open62541 #5721: How to parse UA_ExtensionObject array

zzandyc, 2023-04-04:
I use open62541 Git-Revision: v1.3.5-932-g89a69a8b7

cmake has enabled the following macro:

cmake .. -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_METHODCALLS=ON

when I parse UA_ExtensionObject array, The program crashed,code:
UA_ExtensionObject* arr = (UA_ExtensionObject*)lst.data;
for (size_t idx = 0; idx < lst.arrayLength; ++idx)
{
    UA_Byte* p = (UA_Byte*)arr[idx].content.decoded.data;
    UA_DataTypeMember* member = &arr[idx].content.decoded.type->members[idx];

    UA_Variant val;
    UA_Variant_init(&val);
    
    UA_UInt32 type = member->memberType->typeKind;
    if (member->isArray)
    {
        UA_Variant_setArrayCopy(&val, p, *((size_t*)((UA_Byte*)p + member->padding)), &UA_TYPES[type]);
    }
    UA_Variant_setScalarCopy(&val, (void*)((UA_Byte*)p + member->padding), &UA_TYPES[type]);
}

the path I parsed on the server side is Root->Types->DataTypes->BaseDataTypes->Enumeration->NamingRuleType->EnumValues

jpfr, 2023-04-05:
1. You need an "else" for the non-array case.
2. In order to access members you not only need to consider the padding, but also the size of the member types.
    Better use the `UA_DataType_getStructMember` helper function.
3. The typeKind is not unique. For example, many datatypes are of the "structure" kind.
    Better use `UA_Variant_setScalarCopy(&val, (void*)((UA_Byte*)p + member->padding), member->memberType)`.

zzandyc, 2023-04-07:
Thank you for your reply,I use the follow code to get all attributes for a UA_NodeId, when the UA_NodeId nodeId = UA_NODEID_NUMERIC(0, 12169), the function setVariantValue return false,I don't know where went wrong?
`#include "open62541.h"

bool setVariantValue(void* data, const UA_DataTypeMember* member, UA_Variant* out)
{
    UA_UInt32 type = member->memberType->typeKind;
    if (member->isArray)
    {
        return UA_STATUSCODE_GOOD ==
            UA_Variant_setArrayCopy(out, data, *((size_t*)((UA_Byte*)data + member->padding)), &UA_TYPES[type]);
    }
    return UA_STATUSCODE_GOOD ==
        UA_Variant_setScalarCopy(out, data, &UA_TYPES[type]);
}

void showVal(UA_DataValue* dv)
{
    if (!UA_Variant_isEmpty(&dv->value))
    {
        if (UA_Variant_hasArrayType(&dv->value, dv->value.type))
        {
            if (dv->value.type->typeKind == UA_DATATYPEKIND_EXTENSIONOBJECT)
            {
                bool b = false;
                UA_ExtensionObject* arr = (UA_ExtensionObject*)dv->value.data;
                for (size_t idx = 0; idx < dv->value.arrayLength; ++idx)
                {
                    size_t memberOffset;
                    const UA_DataType* memberType;
                    UA_Boolean isArray;
                    UA_DataTypeMember* member = &arr[idx].content.decoded.type->members[idx];
                    if (!UA_DataType_getStructMember(arr[idx].content.decoded.type, member->memberName,
                        &memberOffset, &memberType, &isArray))
                        continue;

                    void* content = (void*)(((uintptr_t)&arr) + memberOffset);

                    UA_Variant val;
                    UA_Variant_init(&val);
                    b = setVariantValue((void*)arr, member, &val);

                    // use val ...
                }
            }
        }
        else
        {
            //....
        }
    }
}

int main()
{
    UA_Client* client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if (retval != UA_STATUSCODE_GOOD)
    {
        UA_Client_delete(client);
        return (int)retval;
    }

    UA_NodeId nodeId = UA_NODEID_NUMERIC(0, 12169);

    const int arraySize = 27;
    UA_NodeAttributes attrs[arraySize];
    memset(attrs, 0x0, sizeof(attrs));
    for (int i = 0; i < arraySize; ++i)
    {
        attrs[i].specifiedAttributes = UA_ATTRIBUTEID_NODEID + i;
    }

    UA_ReadValueId itemArray[arraySize];
    memset(itemArray, 0x0, sizeof(itemArray));
    for (int i = 0; i < arraySize; ++i)
    {
        UA_ReadValueId_init(&itemArray[i]);
        UA_NodeId_copy(&nodeId, &itemArray[i].nodeId);
        itemArray[i].attributeId = attrs[i].specifiedAttributes;
    }

    UA_ReadRequest request;
    UA_ReadRequest_init(&request);
    request.nodesToRead = itemArray;
    request.nodesToReadSize = arraySize;

    UA_ReadResponse response = UA_Client_Service_read(client, request);
    if (UA_STATUSCODE_GOOD != response.responseHeader.serviceResult)
    {
        return -1;
    }

    if (response.resultsSize != arraySize)
    {
        UA_ReadResponse_clear(&response);
        return -1;
    }

    for (int i = 0; i < response.resultsSize; ++i)
    {
        if (UA_STATUSCODE_GOOD != response.results[i].status)
        {
            continue;
        }

        if (attrs[i].specifiedAttributes == UA_ATTRIBUTEID_VALUE)
        {
            showVal(&response.results[i]);
        }
        else if (attrs[i].specifiedAttributes == UA_ATTRIBUTEID_DATATYPEDEFINITION)
        {
            //showStructureObj(response.results[i].value);
        }
        else
        {
            //showAttr
        }
    }

    UA_Client_delete(client);
    return 0;
}`

closed 2023-04-26 by zzandyc