open62541/open62541 #5731: Parsing nested structs/arrays - OPC-UA Client

rjboer, 2023-04-08:
Framework:
Qt (C++, mingw)

OPC-UA server:
Siemens S7-1500 PLC

In our project we use the open62541 library to setup an opc-ua client. 
The client itself functions without problems, however, we want to parse a nested variable (struct and array within struct) and the system gives out garbage. We use the functions that are included in the client, however examples are scarce and little documented. 

How to parse nested structs with an embedded array as a client following a server registered datatype?

undroidwish, 2023-04-08:
1. Read the DataType of the Variable you want to deserialize.
2. From the DataType NodeId fetch the DataTypeDescription attribute, this gives you a DataTypeDescription with the structure fields.
3. Build a UA_DataType from this description by using information you find in the predefined UA_TYPES array.
4. While building the UA_DataType watch out for types which are still unknown, repeat the steps 2., 3., and 4. recursively until you know all subtypes and have filled an array of UA_DataTypes of all the fancy new types.
5. Hang that UA_DataType array as custom types on your client handle.
6. Now you should be able to serialize/deserialize stuff to/from JSON for example using UA_encodeJson() and UA_decodeJson().

Ensure that your S7 PLC has fresh enough firmware supporting the DataTypeDescription information. Another option would be to parse (hopefully available) XML description which is below /Root/Types/DataTypes/OPC Binary. But this is rather tedious and certainly more complex.

rjboer, 2023-04-08:
Team shall evaluate this on tuesday/wednesday, if it works, PM me your address

phunderr, 2023-04-13:
> 1. Read the DataType of the Variable you want to deserialize.
> 2. From the DataType NodeId fetch the DataTypeDescription attribute, this gives you a DataTypeDescription with the structure fields.
> 3. [...]

I am currently working with this, Fetching the DataTypeDescription or DataTypeDefinition is not possible with the current library so I added the method with the correct attribute ID. The method I implemented is a copy from other read attribute.
static UA_INLINE UA_StatusCode

UA_Client_readDataTypeDefinitionAttribute(UA_Client *client, const UA_NodeId nodeId, UA_DataValue *outDataTypeDefinition) {
    return __UA_Client_readAttribute(client, &nodeId,
                                     UA_ATTRIBUTEID_DATATYPEDEFINITION,
                                     outDataTypeDefinition,
                                     &UA_TYPES[UA_TYPES_DATAVALUE]);
}

Currently the request method is correct and the OPCua server sends the correct data which is confirmed with wireshark, the code cannot interpret this correctly. My guess so far is that the output type is incorrect.

Is DataValue the incorrect type or is there something else that I'm missing

phunderr, 2023-04-14:
When Reading it as a UA_Variant it does not retrieve the data properly the array size is way larger then expected plus the data cannot be read with <invalid address> in the debug screen

[image]

This is the data the PLC sends back after the request is made, this is the data I'm looking for, parsing this data doesn't work

[image]

undroidwish, 2023-04-14:
Well, when it is an `ExtensionObject` why not treat it as such? It could encapsulate the `UA_StructureDefinition`.

phunderr, 2023-04-14:
When i try it with an UA_Extentionobject and cast the data to from content.encoded.body.data it gives an invalid address and the size is way longer then the actual retrieved data.
[image]

undroidwish, 2023-04-14:
In my humble theory the `ExtensionObject` should be decoded (not encoded) and you should see its type in the `content.decoded.type` being `&UA_TYPES[UA_TYPES_STRUCTUREDEFINITION]` and the data pointing to the magically deserialized  `UA_StructureDefinition`. But of course I may be wrong.

phunderr, 2023-04-14:
I think that would be how to read it the problem is that the data is not received the status code im getting is `2147549184` so somewhere in the retrieving of the data it is not read properly

undroidwish, 2023-04-14:
Let's do a cross check. Fetch one of the `vanillatclsh` binaries which match your platform from

  http://www.ch-werner.de/LUCK

Start it from a command shell like so `vanillatclsh... builtin:ckua opc.tcp://your-s7-plc:4840` assuming you can connect unencrypted and w/o user/password. Then navigate in the tree on top to `/Root/Objects/Server/ServerStatus`, press `Tab` and go in the lower list to `DataType`, press `Control-G` to visit the `ServerStatusDataType`. Go again into the list and there select the `DataTypeDefinition` for `ServerStatusDataType` and you should see the full JSON  of the `StructureDefinition` when you press `Control-X`.

Fun fact: If these steps have worked you have used the Tcl binding to open62541 to retrieve the type information, i.e. the same kind of data retrieval which you're currently trying to achieve. And my humble theory holds.

phunderr, 2023-04-14:
I do not think this way works for the PLC I'm working with

[image]

I think the next option is the XML route?

undroidwish, 2023-04-14:
You did not scroll down to the `DataTypeDefinition` attribute of the `ServerStatusDataType` in your screen shot.
We need to know if there's a structure definition.

phunderr, 2023-04-14:
That one has the same 'The attribute is not supported for the specified node' value

undroidwish, 2023-04-14:
This and following threads and references give a different impression:

  https://github.com/FreeOpcUa/opcua-asyncio/issues/234

And what about the firmware status of your S7 PLC?

phunderr, 2023-04-17:
We're using a s7-1512 PLC with the newest V2.9.2 firmware

jpfr, 2023-04-19:
What is the target for decoding? Do you have defined the C structures you want to fill in?

If the type is not defined, then open62541 will provide “opaque” data.
That is, the data is kept in a binary string and wrapped in an ExtensionObject.
The client should never ever provide corrupted data. And so far I don’t have an indication that it does.

phunderr, 2023-04-19:
That is the problem we are having yes, we want to read the datatypes without knowing what the datatypes are, doing this with DataTypeDefinition seems so far not possible. Now i'm going with the XML route by retrieving the xml, parsing it after which the client can be configured so it is readable.

undroidwish, 2023-04-19:
Right now I'm browsing an S7-1518 with the tool I've mentioned earlier (and without having any real idea of the project which is loaded on the PLC). However, I can see non empty DataTypeDefinitions for some user/program defined Structures, but not for the ServerStatusDataType. Thus, I'm somewhat confident that the steps described in my first post point into the right direction.

undroidwish, 2023-04-19:
If you dare to study some Tcl code, the algorithm for making type information is in this procedure:

  https://www.androwish.org/home/file?ci=6f511efd11&name=jni/topcua/library/topcua.tcl&ln=306-475

At the end some `typedef` commands are executed which build up internally an array of `UA_DataType` structs which will be set as `customDataTypes` on the client handle. And from that point on, the client can deal with the glory of the server's data types hopefully.

undroidwish, 2023-05-24:
Meanwhile I've added another helper in my topcua package which generates C code from a live OPC/UA connection to a server or from some XML by using a local inactive server instance. The generated code has an array of `UA_DataType`, a namespace array, and a `UA_NodeId` array of variables obtained from the server, which is to be included into a client application written in at least C.

[The Tcl code is here](https://www.androwish.org/home/file?name=jni/topcua/library/cgen.tcl&ci=tip)

Debatable, how this may help in your case, since it generates static code bound to an interface assumed to be constant.