open62541/open62541 #5738: Nodeset compiler - CS Tightening System - internal type dependency

LaDeKlint, 2023-04-15:
## Description
I am trying to compile the nodeset for the OPC 40451-1: Tightening Systems companion specification. I am guessing it is failing because of an internal type dependency issue that I do not know how to handle. 
## Background Information / Reproduction Steps
Namespaces:
```
http://opcfoundation.org/UA/
http://opcfoundation.org/UA/DI/
http://opcfoundation.org/UA/Machinery/
http://opcfoundation.org/UA/IJT/
```
[nodeset_files.zip](https://github.com/open62541/open62541/files/11240327/nodeset_files.zip)

Used CMake options:
```bash
cmake -DUA_NAMESPACE_ZERO=FULL
```
[CMakeLists.txt](https://github.com/open62541/open62541/files/11240328/CMakeLists.txt):
```cmake
cmake_minimum_required(VERSION 3.2)
project(TightenSystem VERSION 0.1.0)

find_package(open62541 1.3 REQUIRED COMPONENTS FullNamespace)

set(GENERATE_OUTPUT_DIR "${CMAKE_BINARY_DIR}/src_generated/")
file(MAKE_DIRECTORY "${GENERATE_OUTPUT_DIR}")
include_directories("${GENERATE_OUTPUT_DIR}")

# Generate types and namespace for DI
ua_generate_nodeset_and_datatypes(
    NAME "di"
    FILE_CSV "${PROJECT_SOURCE_DIR}/nodeset/DI/Opc.Ua.Di.NodeIds.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/nodeset/DI/Opc.Ua.Di.Types.bsd"
    NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
    FILE_NS "${PROJECT_SOURCE_DIR}/nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
    INTERNAL
)

# Generate types and namespace for Machinery
ua_generate_nodeset_and_datatypes(
    NAME "machinery"
    FILE_NS "${PROJECT_SOURCE_DIR}/nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml"
    DEPENDS "di"
    )

# Generate types and namespace for Tightening
ua_generate_nodeset_and_datatypes(
    NAME "tightening"
    FILE_CSV "${PROJECT_SOURCE_DIR}/nodeset/IJT/Tightening/NodeIds.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/nodeset/IJT/Tightening/Opc.Ua.Ijt.Tightening.NodeSet2.bsd"
    NAMESPACE_MAP "3:http://opcfoundation.org/UA/IJT/"
    FILE_NS "${PROJECT_SOURCE_DIR}/nodeset/IJT/Tightening/Opc.Ua.Ijt.Tightening.NodeSet2.xml"
    DEPENDS "machinery" "di"
    INTERNAL
    )

add_executable(${PROJECT_NAME} 
    ${UA_NODESET_DI_SOURCES}
    ${UA_TYPES_DI_SOURCES}
    ${UA_NODESET_MACHINERY_SOURCES}
    ${UA_NODESET_TIGHTENING_SOURCES}
    ${UA_TYPES_TIGHTENING_SOURCES}
    main.c)

target_link_libraries(${PROJECT_NAME} open62541::open62541)
```
build output:
```bash
[main] Building folder: tighten_system 
[main] The folder containing the CMake cache is missing. The cache will be regenerated.
[main] Configuring project: tighten_system 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -S/home/lars/playground/tighten_system -B/home/lars/playground/tighten_system/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is GNU 9.4.0
[cmake] -- The CXX compiler identification is GNU 9.4.0
[cmake] -- Check for working C compiler: /usr/bin/cc
[cmake] -- Check for working C compiler: /usr/bin/cc -- works
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Check for working CXX compiler: /usr/bin/c++
[cmake] -- Check for working CXX compiler: /usr/bin/c++ -- works
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Found PythonInterp: /usr/bin/python (found version "3.9.5") 
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: /home/lars/playground/tighten_system/build
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/lars/playground/tighten_system/build --config Debug --target all -j 10 --
[build] [ 25%] Generating src_generated/open62541/namespace_di_generated.c, src_generated/open62541/namespace_di_generated.h
[build] [ 25%] Generating src_generated/open62541/namespace_machinery_generated.c, src_generated/open62541/namespace_machinery_generated.h
[build] [ 25%] Generating src_generated/open62541/namespace_tightening_generated.c, src_generated/open62541/namespace_tightening_generated.h
[build] [ 33%] Generating src_generated/open62541/types_tightening_generated.c, src_generated/open62541/types_tightening_generated.h, src_generated/open62541/types_tightening_generated_handling.h
[build] [ 41%] Generating src_generated/open62541/types_di_generated.c, src_generated/open62541/types_di_generated.h, src_generated/open62541/types_di_generated_handling.h
[build] Traceback (most recent call last):
[build]   File "/usr/local/share/open62541/tools/generate_datatypes.py", line 95, in <module>
[build]     parser.create_types()
[build]   File "/usr/local/share/open62541/tools/nodeset_compiler/type_parser.py", line 384, in create_types
[build]     self.parse_types()
[build]   File "/usr/local/share/open62541/tools/nodeset_compiler/type_parser.py", line 426, in parse_types
[build]     self.parseTypeDefinitions(self.outname, f)
[build]   File "/usr/local/share/open62541/tools/nodeset_compiler/type_parser.py", line 328, in parseTypeDefinitions
[build]     raise RuntimeError("Infinite loop detected or type not found while processing types " +
[build] RuntimeError: Infinite loop detected or type not found while processing types TighteningTraceDataType: unknonwn subtype ['tns:StepTraceDataType']. If the unknown subtype is 'Bit', then maybe a struct with optional fields is defined wrong in the .bsd-file. If not, maybe you need to import additional types with the --import flag. E.g. '--import=UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'
[build] make[2]: *** [CMakeFiles/TightenSystem.dir/build.make:128: src_generated/open62541/types_tightening_generated.c] Error 1
[build] make[2]: *** Waiting for unfinished jobs....
[build] INFO:__main__:Preprocessing (existing) /usr/local/share/open62541/tools/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
[build] INFO:__main__:Preprocessing (existing) /usr/local/share/open62541/tools/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
[build] INFO:__main__:Preprocessing (existing) /usr/local/share/open62541/tools/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
[build] INFO:__main__:Preprocessing (existing) /home/lars/playground/tighten_system/nodeset/DI/Opc.Ua.Di.NodeSet2.xml
[build] INFO:__main__:Preprocessing /home/lars/playground/tighten_system/nodeset/DI/Opc.Ua.Di.NodeSet2.xml
[build] INFO:__main__:Preprocessing (existing) /home/lars/playground/tighten_system/nodeset/DI/Opc.Ua.Di.NodeSet2.xml
[build] INFO:__main__:Preprocessing (existing) /home/lars/playground/tighten_system/nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml
[build] INFO:__main__:Preprocessing /home/lars/playground/tighten_system/nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml
[build] INFO:__main__:Skipping Nodeset since it is already loaded: /usr/local/share/open62541/tools/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml 
[build] INFO:__main__:Skipping Nodeset since it is already loaded: /home/lars/playground/tighten_system/nodeset/DI/Opc.Ua.Di.NodeSet2.xml 
[build] INFO:__main__:Preprocessing /home/lars/playground/tighten_system/nodeset/IJT/Tightening/Opc.Ua.Ijt.Tightening.NodeSet2.xml
[build] INFO:__main__:Generating Code for Backend: open62541
[build] INFO:__main__:Generating Code for Backend: open62541
[build] INFO:__main__:NodeSet generation code successfully printed
[build] INFO:__main__:NodeSet generation code successfully printed
[build] INFO:__main__:Generating Code for Backend: open62541
[build] INFO:__main__:NodeSet generation code successfully printed
[build] make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/TightenSystem.dir/all] Error 2
[build] make: *** [Makefile:84: all] Error 2
[proc] The command: /usr/bin/cmake --build /home/lars/playground/tighten_system/build --config Debug --target all -j 10 -- exited with code: 2
[build] Build finished with exit code 2

```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.5
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

closed 2025-02-12 by LaDeKlint

wlkrm, 2025-02-13:
@LaDeKlint, I am having the same problem, have you found the solution?

francisjjp, 2025-03-13:
I have same problem. It can not compile IJT Base, IJT Tightening. (I have used nodeset-injector tool). Error is due to an object.

```
raise RuntimeError("Infinite loop detected or type not found while processing types " +
RuntimeError: Infinite loop detected or type not found while processing types JoiningTraceDataType: unknonwn subtype ['tns:StepTraceDataType']. If the unknown subtype is 'Bit', then maybe a struct with optional fields is defined wrong in the .bsd-file. If not, maybe you need to import additional types with the --import flag. E.g. '--import=UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'
```

wlkrm, 2025-03-13:
@francisjjp, I managed to compile it simply by using the latest version of open62541, if I remember correctly.