open62541/open62541 #5745: server_encryption.c: AccessControl: Unconfigured

growgreat77, 2023-04-18:
## Description
Hi, I am using the "server_encryption.c" example, and I have generated "server_cert.der" and "server_key.der" under the "tools" directory. However, when I run the server, the prompt shows that there is no encryption: "AccessControl: Unconfigured AccessControl. Users have all permissions."Below are the detailed information. What could be the reason for this?
```bash
ysm@ysm:~/opcua1/bin$ ./server ../certs/server_cert.der ../certs/server_key.der ../certs/client_cert.der
[2023-04-18 13:42:43.109 (UTC+0800)] info/eventloop	Starting the EventLoop
[2023-04-18 13:42:43.110 (UTC+0800)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-04-18 13:42:43.110 (UTC+0800)] info/server	AccessControl: Anonymous login is enabled
[2023-04-18 13:42:43.110 (UTC+0800)] info/server	AccessControl: x509 certificate user authentication is enabled
[2023-04-18 13:42:43.110 (UTC+0800)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2023-04-18 13:42:43.110 (UTC+0800)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-04-18 13:42:43.247 (UTC+0800)] warn/userland	ServerUrls already set. Overriding.
[2023-04-18 13:42:43.249 (UTC+0800)] info/securitypolicy	The Basic128Rsa15 security policy with openssl is added.
[2023-04-18 13:42:43.251 (UTC+0800)] info/securitypolicy	The basic256 security policy with openssl is added.
[2023-04-18 13:42:43.252 (UTC+0800)] info/securitypolicy	The basic256sha256 security policy with openssl is added.
[2023-04-18 13:42:43.254 (UTC+0800)] info/securitypolicy	The Aes128Sha256RsaOaep security policy with openssl is added.
[2023-04-18 13:42:43.255 (UTC+0800)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-04-18 13:42:43.255 (UTC+0800)] info/server	AccessControl: Anonymous login is enabled
[2023-04-18 13:42:43.258 (UTC+0800)] info/network	TCP	| Listening on all interfaces
[2023-04-18 13:42:43.258 (UTC+0800)] info/network	TCP 4	| Creating server socket for "0.0.0.0" on port 4840
[2023-04-18 13:42:43.258 (UTC+0800)] info/network	TCP 5	| Creating server socket for "::" on port 4840
[2023-04-18 13:42:43.258 (UTC+0800)] info/network	TCP 6	| Creating server socket for "127.0.1.1" on port 4840
```
This is my file directory.
```bash
├── bin
│   ├── client
│   └── server
├── build
├── certs
│   ├── client_cert.der
│   ├── client_key.der
│   ├── create_self-signed.py
│   ├── localhost.cnf
│   ├── REAMME.MD
│   ├── server_cert.der
│   └── server_key.der
├── CMakeLists.txt
├── open62541
│   ├── open62541.c
│   └── open62541.h
└── src
    ├── client.c
    ├── common.h
    └── server.c
```

This is the encryption certificate information of my server.
```bash
ysm@ysm:~/opcua1/certs$ openssl x509 -in server_cert.der -inform der -noout -text
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            26:81:07:39:ed:7d:2f:0f:b1:ac:be:4f:95:1b:5d:e5:7e:b6:c5:a4
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = DE, L = Here, O = open62541, CN = open62541Server@localhost
        Validity
            Not Before: Apr 18 03:06:50 2023 GMT
            Not After : Apr 17 03:06:50 2024 GMT
        Subject: C = DE, L = Here, O = open62541, CN = open62541Server@localhost
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                Public-Key: (2048 bit)
                Modulus:
                    00:f1:16:01:44:9b:17:5d:0e:84:32:93:08:00:81:
                    c4:09:da:ca:27:a0:88:f3:c3:f2:c6:0b:46:a3:e0:
                    6c:7c:b3:c5:6d:18:75:bb:ee:59:54:5e:a8:55:14:
                    08:08:9d:a1:08:c1:bd:b4:91:2e:bb:95:f0:58:73:
                    3c:4e:67:94:dd:1f:b2:62:c3:c9:e7:9f:54:77:d2:
                    b7:f5:c7:28:aa:6a:5b:26:a3:89:73:a4:80:11:87:
[... 166 lines cut ...]
    if(retval != UA_STATUSCODE_GOOD)
        goto cleanup;

    if(!running)
        goto cleanup; /* received ctrl-c already */
    
    retval = UA_Server_run(server, &running);

 cleanup:
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```
## Checklist

Please provide the following information:

 - [1] open62541 Version (release number or git tag): git clone 2023-04-17,  v1.3.5
 - [1] Other OPC UA SDKs used (client or server):  using Softing OPC client to test 
 - [1] Operating system: ubunt22.04.02 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

growgreat77, 2023-04-18:
And I can indeed log in anonymously using Softing OPC client without any encryption certificates.
[https://industrial.softing.com/products/opc-ua-and-opc-classic-sdks/opc-ua-demo-client.ht](url)

neojaw, 2023-04-18:
Encryption and AccessControl are different things. The Encryption setup and the certificates take care of the encryption of the connection, AccessControl controls what the user can do with an established connection (encrypted or unencrypted).

AccessControl is set up as a plug in that you have to set up yourself to reflect the internal user/permission management of your application. Take a look at the `examples/access_control` folder to get an idea.

As for the unencrypted connection, you are using `UA_ServerConfig_setDefaultWithSecurityPolicies()` which includes the `None` security policy which allows connection without encryption. If you want to prevent that, you have to add the desired security policies manually with the `UA_ServerConfig_addSecurityPolicy*()` functions instead.

growgreat77, 2023-04-18:
> Encryption and AccessControl are different things. The Encryption setup and the certificates take care of the encryption of the connection, AccessControl controls what the user can do with an established connection (encrypted or unencrypted).
> 
> AccessControl is realized as a plug in that you have to set up yourself to reflect the internal user/permission management of your application. Take a look at the `examples/access_control` folder to get an idea.
> 
> As for the unencrypted connection, you are using `UA_ServerConfig_setDefaultWithSecurityPolicies()` which includes the `None` security policy which allows connection without encryption. If you want to prevent that, you have to add the desired security policies manually with the `UA_ServerConfig_addSecurityPolicy*()` functions instead.
> 
> Edit: Note that instead of excluding the `None` policy completely, you might want to restrict it for _discovery_ only, by setting `config->securityPolicyNoneDiscoveryOnly = true;`.

Thank you for helping me. My main goal is only to allow connection with encryption, so I modified the code.  my updated code is :
```bash
    UA_StatusCode retval = UA_ServerConfig_addSecurityPolicyBasic256Sha256(config, &certificate, &privateKey);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_LOG_WARNING(&config->logger, UA_LOGCATEGORY_USERLAND,
                       "Could not add SecurityPolicy#Basic256Sha256 with error code %s",
                       UA_StatusCode_name(retval));
    }

```
but the prompts is still no encrypting SecurityPolicy:
```bash
Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
```
Furthermore, I can still connect to this server without any encryption or anonymously useing this client [:](url)[https://industrial.softing.com/products/opc-ua-and-opc-classic-sdks/opc-ua-demo-client.ht](https://github.com/open62541/open62541/issues/url).

neojaw, 2023-04-18:
Do you still have the `UA_ServerConfig_setDefaultWithSecurityPolicies()` call? Because that log line comes from `UA_AccessControl_default()` which is called from there and I don't see you doing it explicitly. This would also explain the unencrypted connection.

growgreat77, 2023-04-18:
> Do you still have the `UA_ServerConfig_setDefaultWithSecurityPolicies()` call? Because that log line comes from `UA_AccessControl_default()` which is called from there and I don't see you doing it explicitly. This would also explain the unencrypted connection.

I add the desired security policies manually with the UA_ServerConfig_addSecurityPolicy*() functions instead of UA_ServerConfig_setDefaultWithSecurityPolicies()
```bash
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information.
 *
 *    Copyright 2019 (c) Kalycito Infotech Private Limited
 *    Copyright 2021 (c) Christian von Arnim, ISW University of Stuttgart (for VDW and umati)
 *
 */

#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/plugin/create_certificate.h>
#include <open62541/plugin/securitypolicy.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

#include "common.h"

UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(int argc, char* argv[]) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);
    UA_ByteString certificate = UA_BYTESTRING_NULL;
    UA_ByteString privateKey = UA_BYTESTRING_NULL;
    if(argc >= 3) {
        /* Load certificate and private key */
        certificate = loadFile(argv[1]);
        privateKey = loadFile(argv[2]);
    } else {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "Missing arguments. Arguments are "
                     "<server-certificate.der> <private-key.der> "
                     "[<trustlist1.crl>, ...]");
#if defined(UA_ENABLE_ENCRYPTION_OPENSSL) || defined(UA_ENABLE_ENCRYPTION_LIBRESSL)
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                    "Trying to create a certificate.");
        UA_String subject[3] = {UA_STRING_STATIC("C=DE"),
                            UA_STRING_STATIC("O=SampleOrganization"),
                            UA_STRING_STATIC("CN=Open62541Server@localhost")};
        UA_UInt32 lenSubject = 3;
        UA_String subjectAltName[2]= {
            UA_STRING_STATIC("DNS:localhost"),
            UA_STRING_STATIC("URI:urn:open62541.server.application")
        };
        UA_UInt32 lenSubjectAltName = 2;
        UA_StatusCode statusCertGen =
            UA_CreateCertificate(UA_Log_Stdout,
                                 subject, lenSubject,
                                 subjectAltName, lenSubjectAltName,
                                 0, UA_CERTIFICATEFORMAT_DER,
                                 &privateKey, &certificate);

        if(statusCertGen != UA_STATUSCODE_GOOD) {
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "Generating Certificate failed: %s",
                UA_StatusCode_name(statusCertGen));
            return EXIT_SUCCESS;
        }
#else
        return EXIT_SUCCESS;
#endif
    }

    /* Load the trustlist */
    size_t trustListSize = 0;
    if(argc > 3)
        trustListSize = (size_t)argc-3;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize+1);
    for(size_t i = 0; i < trustListSize; i++)
        trustList[i] = loadFile(argv[i+3]);

    /* Loading of an issuer list, not used in this application */
    size_t issuerListSize = 0;
    UA_ByteString *issuerList = NULL;

    /* Loading of a revocation list currently unsupported */
    UA_ByteString *revocationList = NULL;
    size_t revocationListSize = 0;

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    
    UA_StatusCode retval = UA_ServerConfig_addSecurityPolicyBasic256Sha256(config, &certificate, &privateKey);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_LOG_WARNING(&config->logger, UA_LOGCATEGORY_USERLAND,
                       "Could not add SecurityPolicy#Basic256Sha256 with error code %s",
                       UA_StatusCode_name(retval));
    }

    // UA_StatusCode retval =
    //     UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
    //                                                    &certificate, &privateKey,
    //                                                    trustList, trustListSize,
    //                                                    issuerList, issuerListSize,
    //                                                    revocationList, revocationListSize);

    #ifdef UA_ENABLE_WEBSOCKET_SERVER
    UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(server), 7681, 0, 0, &certificate, &privateKey);
    #endif

    UA_ByteString_clear(&certificate);
    UA_ByteString_clear(&privateKey);
    for(size_t i = 0; i < trustListSize; i++)
        UA_ByteString_clear(&trustList[i]);
    if(retval != UA_STATUSCODE_GOOD)
        goto cleanup;

    if(!running)
        goto cleanup; /* received ctrl-c already */
    
    retval = UA_Server_run(server, &running);

 cleanup:
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

neojaw, 2023-04-18:
Did you recompile? Because that code does not even run on my machine since there is no CertificateVerification set up.

```
UA_CertificateVerification_Trustlist(&config->certificateVerification, trustList, trustListSize, issuerList, issuerListSize, revocationList, revocationListSize);
```

growgreat77, 2023-04-18:
> Did you recompile? Because that code does not even run on my machine since there is no CertificateVerification set up.
> 
> ```
> UA_CertificateVerification_Trustlist(&config->certificateVerification, trustList, trustListSize, issuerList, issuerListSize, revocationList, revocationListSize);
> ```

Did you put the common.h in directory or generate key? I deleted all the previously compiled files and then made some modifications to the function UA_ServerConfig_addSecurityPolicyBasic256Sha256 you mentioned. Then I went to the build directory, ran "cmake .. && make". Here is my modified CMakeLists.txt code:
```bash
cmake_minimum_required(VERSION 3.5)

project(OPCUA1)

set (EXECUTABLE_OUTPUT_PATH  ${PROJECT_SOURCE_DIR}/bin)

add_definitions(-std=c99)

include_directories(${PROJECT_SOURCE_DIR}/open62541)
include_directories(${PROJECT_SOURCE_DIR}/src)

find_package(OpenSSL REQUIRED)

add_executable(server ${PROJECT_SOURCE_DIR}/src/server.c ${PROJECT_SOURCE_DIR}/open62541/open62541.c)
target_link_libraries(server ${OPENSSL_LIBRARIES})

add_executable(client ${PROJECT_SOURCE_DIR}/src/client.c ${PROJECT_SOURCE_DIR}/open62541/open62541.c)
target_link_libraries(client ${OPENSSL_LIBRARIES})
```

growgreat77, 2023-04-18:
> Encryption

everythin is going well in my tutorials, except the Encryption. I get the code with git clone.

ayarikba, 2024-11-20:
Did you solve the problem you mentioned ?