open62541/open62541 #5749: open62541 server crash when browsing its nodesets with UaExpert

danmoser, 2023-04-19:
## Description
UaExpert [website](https://www.unified-automation.com/products/development-tools/uaexpert.html) has a tool to export server nodesets.

When trying to do so with **any of the open62541 example servers, they crash** without an specific error message (see log below).

## Background Information / Reproduction Steps
- start an example server
- open UaExpert > add new document > "XML Nodeset Export"
- Export selected Namespaces > "1:urn:open62541.server.application"
- (server crashes)

Log with 50 level doesn't provide much (currently https://paste2.org/ is offline)
```
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 0 | Session "Administrator" | Read attribute 3 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 14 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 15 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 16 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 17 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 18 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 19 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 20 of Node ns=1;i=53796
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 3 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 4 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 5 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 6 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 7 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 13 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 0 | Session "Administrator" | Read attribute 3 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 14 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 15 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 16 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 17 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 18 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 19 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 20 of Node ns=1;i=53797
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 3 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 4 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 5 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 6 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 7 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 2 | Session "urn:orion:UnifiedAutomation:UaExpert" | Read attribute 13 of Node ns=1;i=53798
[2023-04-19 09:03:46.563 (UTC-0600)] debug/session[0m	SecureChannel 0 | Session "Administrator" | Read attribute 3 of Node ns=1;i=53798
Segmentation fault (core dumped)
```

Used CMake options:
```
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/user/home/.local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=MBEDTLS
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=50
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3
 - [ ] Other OPC UA SDKs used (client or server): UaExpert (client)
 - [X] Operating system: Linux (RHEL8)
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-04-21:
Hmm. I cannot reproduce with version v1.3.5 and the server_ctt example.
Valgrind also doesn't complain.
Can you be more specific about the version and circumstances so we can reproduce?

```bash
jpfr@virtualbox:~/software/open62541/build$ valgrind ./bin/examples/server_ctt ./server_cert.der ./server_key.der --enableAnonymous --enableUnencrypted
==4179== Memcheck, a memory error detector
==4179== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==4179== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==4179== Command: ./bin/examples/server_ctt ./server_cert.der ./server_key.der --enableAnonymous --enableUnencrypted
==4179==
[2023-04-21 10:59:16.533 (UTC+0200)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-04-21 10:59:16.547 (UTC+0200)] info/server        AccessControl: Anonymous login is enabled
[2023-04-21 10:59:17.249 (UTC+0200)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=integer): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-04-21 10:59:17.252 (UTC+0200)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=uinteger): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-04-21 10:59:17.761 (UTC+0200)] info/network       TCP network layer listening on opc.tcp://virtualbox:4840/
[2023-04-21 10:59:19.945 (UTC+0200)] info/network       Connection 5 | New connection over TCP from 192.168.56.1
[2023-04-21 10:59:19.980 (UTC+0200)] info/channel       Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-04-21 10:59:20.023 (UTC+0200)] info/channel       Connection 5 | SecureChannel 1 | CloseSecureChannel
[2023-04-21 10:59:20.028 (UTC+0200)] info/network       Connection 5 | Closed
[2023-04-21 10:59:21.753 (UTC+0200)] info/network       Connection 5 | New connection over TCP from 192.168.56.1
[2023-04-21 10:59:21.755 (UTC+0200)] info/channel       Connection 5 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-04-21 10:59:21.767 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | Session created
[2023-04-21 10:59:21.787 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | ActivateSession: Session activated
[2023-04-21 10:59:21.825 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | Subscription 1 | Subscription created (Publishing interval 100.00ms, max 1000 notifications per publish)
[2023-04-21 10:59:21.836 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | Subscription 1 | Could not create a MonitoredItem with StatusCode BadNodeIdInvalid
[2023-04-21 10:59:21.841 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | Subscription 1 | Subscription deleted
[2023-04-21 10:59:29.426 (UTC+0200)] info/session       SecureChannel 2 | Session "urn:KIS-jul32602-1:UnifiedAutomation:UaExpert" | Closing the Session
[2023-04-21 10:59:29.430 (UTC+0200)] info/channel       Connection 5 | SecureChannel 2 | CloseSecureChannel
[2023-04-21 10:59:29.430 (UTC+0200)] info/network       Connection 5 | Closed
^C[2023-04-21 10:59:31.116 (UTC+0200)] info/server      Received Ctrl-C
[2023-04-21 10:59:31.121 (UTC+0200)] warn/server        Shutting down the server with a delay of 5000 ms
[2023-04-21 10:59:36.144 (UTC+0200)] info/network       Shutting down the TCP network layer
==4179==
==4179== HEAP SUMMARY:
==4179==     in use at exit: 0 bytes in 0 blocks
==4179==   total heap usage: 62,550 allocs, 62,550 frees, 63,426,868 bytes allocated
==4179==
==4179== All heap blocks were freed -- no leaks are possible
==4179==
==4179== For lists of detected and suppressed errors, rerun with: -s
==4179== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

danmoser, 2023-04-21:
Thank you @jpfr for the prompt response. I believe its something related to RedHat linux OS... Using RHEL 7 (instead of RHEL8), we got the error below. However, using an Ubuntu 22.04 VM, I could export the nodesets!!!

```
#0  0x00007ffff7659725 in __memcpy_ssse3_back () from /lib64/libc.so.6
#1  0x00000000008bb347 in UA_Array_copy (src=0x9, size=14126544, dst=0xd78e78, type=0x9d07b0 <UA_TYPES+144>) at /export/home/user/proj/open62541/src/ua_types.c:1764
#2  0x00000000008b6c2c in String_copy (src=0x7fffffffce80, dst=0xd78e70, _=0x9d0a38 <UA_TYPES+792>) at /export/home/user/proj/open62541/src/ua_types.c:155
#3  0x00000000008b975e in UA_copy (src=0x7fffffffce80, dst=0xd78e70, type=0x9d0a38 <UA_TYPES+792>) at /export/home/user/proj/open62541/src/ua_types.c:1198
#4  0x00000000008bb38c in UA_Array_copy (src=0x7fffffffce80, size=1, dst=0x7fffffffd0b8, type=0x9d0a38 <UA_TYPES+792>) at /export/home/user/proj/open62541/src/ua_types.c:1772
#5  0x00000000008b7d5d in UA_Variant_setArrayCopy (v=0x7fffffffd0a0, array=0x7fffffffce80, arraySize=1, type=0x9d0a38 <UA_TYPES+792>) at /export/home/user/proj/open62541/src/ua_types.c:605
#6  0x000000000040b194 in readSessionDiagnostics (server=0x9e1440, sessionId=0xc16f80, sessionContext=0x0, nodeId=0xc02130, nodeContext=0x0, sourceTimestamp=false, range=0x0, value=0x7fffffffd0a0)
    at /export/home/user/proj/open62541/src/server/ua_server_ns0_diagnostics.c:403
#7  0x000000000041e46d in readValueAttributeFromDataSource (server=0x9e1440, session=0xc16eb8, vn=0xc02130, v=0xd5a170, timestamps=UA_TIMESTAMPSTORETURN_NEITHER, rangeptr=0x0)
    at /export/home/user/proj/open62541/src/server/ua_services_attribute.c:163
#8  0x000000000041e766 in readValueAttributeComplete (server=0x9e1440, session=0xc16eb8, vn=0xc02130, timestamps=UA_TIMESTAMPSTORETURN_NEITHER, indexRange=0xd2e9f8, v=0xd5a170)
    at /export/home/user/proj/open62541/src/server/ua_services_attribute.c:229
#9  0x000000000041f1f7 in ReadWithNode (node=0xc02130, server=0x9e1440, session=0xc16eb8, timestampsToReturn=UA_TIMESTAMPSTORETURN_NEITHER, id=0xd2e9d8, v=0xd5a170)
    at /export/home/user/proj/open62541/src/server/ua_services_attribute.c:446
#10 0x000000000041f720 in Operation_Read (server=0x9e1440, session=0xc16eb8, request=0x7fffffffd9b0, rvi=0xd2e9d8, result=0xd5a170)
    at /export/home/user/proj/open62541/src/server/ua_services_attribute.c:571
#11 0x000000000040ccd5 in UA_Server_processServiceOperations (server=0x9e1440, session=0xc16eb8, operationCallback=0x41f6aa <Operation_Read>, context=0x7fffffffd9b0, requestOperations=0x7fffffffda30, 
    requestOperationsType=0x9d46f8 <UA_TYPES+16344>, responseOperations=0x7fffffffd908, responseOperationsType=0x9d0d50 <UA_TYPES+1584>)
    at /export/home/user/proj/open62541/src/server/ua_server_utils.c:330
#12 0x000000000041f835 in Service_Read (server=0x9e1440, session=0xc16eb8, request=0x7fffffffd9b0, response=0x7fffffffd880) at /export/home/user/proj/open62541/src/server/ua_services_attribute.c:607
#13 0x00000000008cf071 in processMSGDecoded (server=0x9e1440, channel=0xb40998, requestId=668, service=0x41f75f <Service_Read>, request=0x7fffffffd9b0, requestType=0x9d4740 <UA_TYPES+16416>, 
    response=0x7fffffffd880, responseType=0x9d4788 <UA_TYPES+16488>, sessionRequired=true, counterOffset=264) at /export/home/user/proj/open62541/src/server/ua_server_binary.c:703
#14 0x00000000008cf67b in processMSG (server=0x9e1440, channel=0xb40998, requestId=668, msg=0xbfaa88) at /export/home/user/proj/open62541/src/server/ua_server_binary.c:816
#15 0x00000000008cf896 in processSecureChannelMessage (application=0x9e1440, channel=0xb40998, messagetype=UA_MESSAGETYPE_MSG, requestId=668, message=0xbfaa88)
    at /export/home/user/proj/open62541/src/server/ua_server_binary.c:844
#16 0x00000000008c73be in assembleProcessMessage (channel=0xb40998, application=0x9e1440, callback=0x8cf6d3 <processSecureChannelMessage>)
    at /export/home/user/proj/open62541/src/ua_securechannel.c:634
#17 0x00000000008c7a79 in processChunks (channel=0xb40998, application=0x9e1440, callback=0x8cf6d3 <processSecureChannelMessage>) at /export/home/user/proj/open62541/src/ua_securechannel.c:781
#18 0x00000000008c7ef7 in UA_SecureChannel_processBuffer (channel=0xb40998, application=0x9e1440, callback=0x8cf6d3 <processSecureChannelMessage>, buffer=0x7fffffffdd70)
    at /export/home/user/proj/open62541/src/ua_securechannel.c:897
#19 0x00000000008cfbe0 in UA_Server_processBinaryMessage (server=0x9e1440, connection=0xd03670, message=0x7fffffffdd70) at /export/home/user/proj/open62541/src/server/ua_server_binary.c:914
#20 0x00000000008b5579 in ServerNetworkLayerTCP_listen (nl=0xa43d60, server=0x9e1440, timeout=0) at /export/home/user/proj/open62541/arch/network_tcp.c:577
#21 0x00000000004043d7 in UA_Server_run_iterate (server=0x9e1440, waitInternal=false) at /export/home/user/proj/open62541/src/server/ua_server.c:709
#22 0x0000000000402686 in main (argc=1, argv=0x7fffffffe098) at /export/home/user/proj/open62541/examples/server_mainloop.c:45
```

jpfr, 2023-04-25:
That is strange. Anyway we should resolve it on RHEL7 as well.
To make sure we understand the origin of this.

Would you please reproduce the crash with valgrind to get a more thorough debug output?
Please use `--exit-on-first-error=yes` so that we don't have memory access errors buried in the log output.

danmoser, 2023-04-26:
Julius, this is what I got:
```
valgrind --exit-on-first-error=yes ./server
==3119569== Bad option: --exit-on-first-error=yes
==3119569== You must define a non nul exit error code, with --error-exitcode=...
==3119569== Use --help for more information or consult the user manual.
==3119569== Memcheck, a memory error detector
==3119569== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==3119569== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==3119569== Command: ./server
==3119569== 
[2023-04-26 10:29:21.315 (UTC-0600)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-04-26 10:29:21.332 (UTC-0600)] info/server	AccessControl: Anonymous login is enabled
[2023-04-26 10:29:21.332 (UTC-0600)] info/server	AccessControl: x509 certificate user authentication is enabled
[2023-04-26 10:29:21.334 (UTC-0600)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2023-04-26 10:29:21.336 (UTC-0600)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-04-26 10:29:21.344 (UTC-0600)] info/session	SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=the.answer): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2023-04-26 10:29:21.375 (UTC-0600)] info/network	TCP network layer listening on opc.tcp://orion:4840/
[2023-04-26 10:29:37.642 (UTC-0600)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2023-04-26 10:29:37.678 (UTC-0600)] info/channel	Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-04-26 10:29:37.719 (UTC-0600)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2023-04-26 10:29:37.723 (UTC-0600)] info/network	Connection 5 | Closed
[2023-04-26 10:29:38.614 (UTC-0600)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2023-04-26 10:29:38.615 (UTC-0600)] info/channel	Connection 5 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-04-26 10:29:38.618 (UTC-0600)] info/channel	Connection 5 | SecureChannel 2 | CloseSecureChannel
[2023-04-26 10:29:38.619 (UTC-0600)] info/network	Connection 5 | Closed
[2023-04-26 10:29:40.365 (UTC-0600)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2023-04-26 10:29:40.366 (UTC-0600)] info/channel	Connection 5 | SecureChannel 3 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-04-26 10:29:40.367 (UTC-0600)] info/channel	Connection 5 | SecureChannel 3 | CloseSecureChannel
[2023-04-26 10:29:40.367 (UTC-0600)] info/network	Connection 5 | Closed
[2023-04-26 10:29:40.431 (UTC-0600)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2023-04-26 10:29:40.432 (UTC-0600)] info/channel	Connection 5 | SecureChannel 4 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-04-26 10:29:40.517 (UTC-0600)] info/session	SecureChannel 4 | Session "urn:orion:UnifiedAutomation:UaExpert" | Session created
[2023-04-26 10:29:40.528 (UTC-0600)] info/session	SecureChannel 4 | Session "urn:orion:UnifiedAutomation:UaExpert" | ActivateSession: Session activated
[2023-04-26 10:29:40.585 (UTC-0600)] info/session	SecureChannel 4 | Session "urn:orion:UnifiedAutomation:UaExpert" | Subscription 1 | Subscription created (Publishing interval 100.00ms, max 1000 notifications per publish)
[2023-04-26 10:29:40.614 (UTC-0600)] info/session	SecureChannel 4 | Session "urn:orion:UnifiedAutomation:UaExpert" | Subscription 1 | MonitoredItem 1 | Created the MonitoredItem (Sampling Interval: 50.00ms, Queue Size: 100)
==3119569== Invalid read of size 1
==3119569==    at 0x4C41330: memmove (vg_replace_strmem.c:1382)
==3119569==    by 0x4E69BB9: UA_Array_copy (ua_types.c:1764)
==3119569==    by 0x4E6549C: String_copy (ua_types.c:155)
==3119569==    by 0x4E67FEB: UA_copy (ua_types.c:1198)
==3119569==    by 0x4E69BFE: UA_Array_copy (ua_types.c:1772)
==3119569==    by 0x4E665CE: UA_Variant_setArrayCopy (ua_types.c:605)
==3119569==    by 0x4E8527F: readSessionDiagnostics (ua_server_ns0_diagnostics.c:403)
==3119569==    by 0x4EA5D6B: readValueAttributeFromDataSource (ua_services_attribute.c:163)
==3119569==    by 0x4EA6052: readValueAttributeComplete (ua_services_attribute.c:229)
==3119569==    by 0x4EA6B22: ReadWithNode (ua_services_attribute.c:446)
==3119569==    by 0x4EA7082: Operation_Read (ua_services_attribute.c:571)
==3119569==    by 0x4E89CD3: UA_Server_processServiceOperations (ua_server_utils.c:330)
==3119569==  Address 0x9 is not stack'd, malloc'd or (recently) free'd
==3119569== 
==3119569== 
==3119569== Exit program on first error (--exit-on-first-error=yes)
```