open62541/open62541 #5759: How to achieve a samplinginterval less then 5ms ?

ThWulf, 2023-04-25:
## The Question

In `server.h` it reads:
```
UA_DurationRange samplingIntervalLimits; /* in ms (must not be less than 5) */
```

1. Why is 5ms seconds the limit and is there a way to achieve lower sampling rates ? (e.g. by tweaking a compile time constant?)

## The Intention and ideas for a workaround

I would like to sample some values with a rate of 2ms (at best effort). In case there is no way to achieve lower sampling rates, I had the idea of using the historizing mechanism. Thus, I have to other questions:

2. Is it possible to set a polling rate lower then 5ms ? There is no such limitation mentioned in `plugins\include\open62541\plugin\historydata\history_data_gathering.h` 

3. Is it possible to set the historizing strategy to `UA_HISTORIZINGUPDATESTRATEGY_USER` and to have custom thread, filling the history data at a rate of 2ms ?

4. Is it possible to achieve sampling rates of 2ms with the pubSub-Mechanism ?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.3.1

## Related Issues
* #2106 (There was only a check added to prevent users running into it unknowingly)

jpfr, 2023-06-02:
You can even set the sampling rate to zero.
Then we sample every time the variable is written.

jpfr, 2023-06-02:
The configuration is here:

https://github.com/open62541/open62541/blob/2d3e1b06165c893bbb275e8bb6d579bc33663f41/include/open62541/server.h#L307

We could allow lower sampling intervals and only issue a warning instead.

NoelGraf, 2023-07-04:
@ThWulf can be closed?

closed 2023-07-25 by NoelGraf