open62541/open62541 #5765: cc1.exe terminal remain opened when running C in windows10

AmAwais2242, 2023-04-29:
Hi:
I'm trying to run my first program in C, vs code, windows 10, but whenever i run the file a terminal opens and it remain open for 5 to 10 minutes, it is a blank terminal and shows path (C:\mingw\libexec\gcc\mingw32\6.3.0\gcc1.exe) it is an blank terminal, after few time my vs code terminal shows me this error

c:/mingw/bin/../lib/gcc/mingw32/6.3.0/../../../libmingw32.a(main.o):(.text.startup+0xa0): undefined reference to `WinMain@16'
collect2.exe: error: ld returned 1 exit status

I'm just trying to compile basic 'Hello world' program, path to environment variables already defined. an image of the error is also attached
[image]
 This image shows about environment variables

[image]

I have also tried to run file using codeblock, but same thing happens.

kindly help me I want to learn C.

Farouk8545, 2024-07-02:
please let me know if you found a solution because i having the same issue

Vilbrequin, 2024-07-03:
try to run vs code as an administrator, i had the same problem and it works when i run vs code as admin.

Vilbrequin, 2024-07-03:
@AmAwais2242 @Farouk8545 
try to run vs code as an administrator, i had the same problem and it works when i run vs code as admin.

Farouk8545, 2024-07-03:
Turned out it was my antivirus, but thanks man

Sudhanshu7808, 2024-10-19:
Same issue with me . Please agar kisi ko mila hai solution to batao.