open62541/open62541 #5772: Error UA_STATUSCODE_BADCERTIFICATEURIINVALID when using Certificate

ventil99, 2023-05-06:
Hello everyone, I'm currently trying to implement an OPC client based on the Open62541 project and maybe I could need a little help or information.

I tried to connect with KEP server (version 6.5.829.0), but couldn't get it to work. I received the error code UA_STATUSCODE_BADCERTIFICATEURIINVALID.

Connecting without the usage of any certificate works fine.

The certificate I use is also being used by some client based on another OPC UA implementation (Unified Automation), which is able to connect to the KEP server.

 

Could you please tell me which variables has to be set within the source code, so I might be able to connect to the KEP server with the open62541 client?

The Code and the Zertificate i used are in the .zip-File.
[Certificate_Code.zip](https://github.com/open62541/open62541/files/11412511/Certificate_Code.zip)

Open62541. lib was build using Ninja and bcc32c.exe, OpenSSL was set for Encryption.
open62541 version: 1.3
OpenSLL version: 1.1.1.2
IDE: RAD Studio 10.4

ccvca, 2023-05-10:
`cc->clientDescription.applicationUri = ... ` propably need to be set after `UA_ClientConfig_setDefaultEncryption`, as this function will set the default uri (if I remember this correctly).

ventil99, 2023-05-11:
> `cc->clientDescription.applicationUri = ... ` propably need to be set after `UA_ClientConfig_setDefaultEncryption`, as this function will set the default uri (if I remember this correctly).

Thanks for your suggestion, I tried it, but it doesn't work, I get the same Errorcode

Alexk12, 2023-05-21:
// Replace:
    cc->clientDescription.applicationUri = UA_STRING_ALLOC("urn:VMSUBTILW10RAD:LOTASSoftwareentwicklung:OPC_UA_Client, DNS:VMSUBTILW10RAD");

    // With:
    UA_String_clear(&cc->clientDescription.applicationUri);
    cc->clientDescription.applicationUri = UA_String_fromChars("urn:VMSUBTILW10RAD:LOTASSoftwareentwicklung:OPC_UA_Client");

Alexk12, 2023-05-21:
> `cc->clientDescription.applicationUri = ... ` propably need to be set after `UA_ClientConfig_setDefaultEncryption`, as this function will set the default uri (if I remember this correctly).

Setting application Uri can/should be placed before setting default encryption. UA_ClientConfig_setDefaultEncryption function just adds security policies supported by the client and it does not use and does not change application Uri.

NoelGraf, 2023-07-04:
@ventil99 What is the current status?

NoelGraf, 2023-07-25:
Close the issue for now, reopen it if necessary.

closed 2023-07-25 by NoelGraf