open62541/open62541 #5782: UA_SECURECHANNELSTATE_OPN_SENT is missing in connectActivity

Alexk12, 2023-05-13:
## Description
It seems UA_SECURECHANNELSTATE_OPN_SENT state is missing in connectActivity functin (clent code).
This results in closing security channel during connection (with BadInternalError result).
The issue occurred at connection to server with security, when server rejected client's certificate. (Depends on server.)

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

peschuster, 2024-05-28:
We are having the same issue with a client based on open62541 after an update from v1.3 to v1.4.
The client program works fine with open62541 v1.3.x

With the fix mentioned above the client program works also with open62541 v1.4.1.

It could be related to the following commit that is only in v1.4: 99f56d355220151c4608946d1f0828b374c92e49
Any ideas @jpfr? Is the fix mentioned above viable or is the root cause somewhere else?

The server implementation is based on open62541 v1.3.x
The problem only occurs with this open62541-based OPC UA server. Other OPC UA server implementations (3rd party) do not cause this problem. So this could also be an issue in our open62541 server integration. But I have no idea where to look. Maybe our embedded server is just taking too long to answer the request?

o6-nbeier, 2026-10-20:
Thank you for pointing this out.

The missing case was added in #6513, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier