open62541/open62541 #5783: Cliend hangs if server rejects client's sertificate and sends error to client

Alexk12, 2023-05-14:
## Description

Conditions:
1. Client connects with security.
2. Server rejects client's certificate and sends error.

If error received on early stage when channel not opened yet (for example, if server rejects client's certificate and reports error), setting state to UA_SECURECHANNELSTATE_CLOSING prevents exit from event loop (client hangs).

Suggested fix:

    void
    processERRResponse(UA_Client *client, const UA_ByteString *chunk) {
        //client->channel.state = UA_SECURECHANNELSTATE_CLOSING; // FIX: commented because this state is set below on call to closeSecureChannel

        size_t offset = 0;
        UA_TcpErrorMessage errMessage;
        UA_StatusCode res =
            UA_decodeBinaryInternal(chunk, &offset, &errMessage,
                                    &UA_TRANSPORT[UA_TRANSPORT_TCPERRORMESSAGE], NULL);
        if(res != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR_CHANNEL(&client->config.logger, &client->channel,
                                 "Received an ERR response that could not be decoded "
                                 "with StatusCode %s", UA_StatusCode_name(res));
            client->connectStatus = res;
            closeSecureChannel(client);
            return;
        }
    
        UA_LOG_ERROR_CHANNEL(&client->config.logger, &client->channel,
                             "Received an ERR response with StatusCode %s and the following "
                             "reason: %.*s", UA_StatusCode_name(errMessage.error),
                             (int)errMessage.reason.length, errMessage.reason.data);
        client->connectStatus = errMessage.error;
        closeSecureChannel(client);
        UA_TcpErrorMessage_clear(&errMessage);
    }

## Background Information / Reproduction Steps

Used CMake options:
```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

peschuster, 2024-06-12:
We also encountered this bug.

It was probably caused by the added "recursion check" in `closeSecureChannel` (commit 4bd6f4023ab59fa8ba1f17f5d82dd89636fc9995) that was added in v1.4.0-rc1

`UA_SecureChannel_shutdown` already sets `UA_SECURECHANNELSTATE_CLOSING` but is never called when the state is already set to "`..._CLOSING`". So the suggested fix seems legit to me.

o6-nbeier, 2026-10-20:
Thank you for the report.

The channel state handling was corrected in #6541, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier