open62541/open62541 #5784: Need advice for integration into an event loop

paddor, 2023-05-15:
I've integrated the server into the event loop of my framework using `UA_Server_run_iterate()`. To keep CPU usage low despite a lot of variables, I call the function only once every 50ms, but that does not seem to be enough with long subscriptions. What's the best way for to adapt the interval based on load? Is there any way to determine the optimal interval between calls to `UA_Server_run_iterate()`, like a queue length? The return value of `UA_Server_run_iterate()` only indicates how long to wait until the next scheduled callback. It doesn't take load into consideration.

undroidwish, 2023-05-19:
My advise in your case is to go the multi threaded path, i.e. have a dedicated thread serving OPC/UA and let the rest of your framework perform its usual tasks. For the (hopefully few) shared resources between the dedicated thread and the framework you'll have to invent some IPC and/or locking scheme to ensure data consistency. But performance wise it's the best approach.

paddor, 2024-02-19:
Thanks for the advice. For the moment we want to stay away from threaded approaches in our event-driven, single-threaded, multi-process architecture. As a compromise we call `UA_Server_run_iterate()` 10 times every 30ms, which seems to work well in our use case and does not use a lot of CPU.

I'm keeping this issue open for now. Hopefully more tools for introspection are implemented some day which would allow a more adaptive solution.

closed 2024-07-16 by paddor