open62541/open62541 #5787: There is a memory leak problem in m1

0x00500, 2023-05-18:
`#include <string>
#include "open62541.h"

class TEST
{
public:
    TEST(std::string url)
    {
        m_status_ = UA_STATUSCODE_BAD;
        m_client_ = nullptr;
        m_url_ = url;
    }

    void connectService() {
        if (m_status_ == UA_STATUSCODE_GOOD) return;
        if (m_client_) {
            UA_Client_disconnect(m_client_);
            UA_Client_delete(m_client_);
            m_client_ = nullptr;
        }

        m_client_ = UA_Client_new();
        UA_ClientConfig_setDefault(UA_Client_getConfig(m_client_));
        // UA_Variant_init(&m_value_);

        while (true) {
            m_status_ = UA_Client_connect(m_client_, m_url_.c_str());
            if (m_status_ == UA_STATUSCODE_GOOD) break;
            UA_sleep_ms(100);
        }
    }

private:
    UA_StatusCode m_status_;
    UA_Client * m_client_;
    std::string m_url_;

};

int main(void)
{
    TEST obj("opc.tcp://192.168.30.250:30600");
    obj.connectService();

    return 0;
}`

Git-Revision: v1.3.5-undefined
lib-open62541.h||open62541.c
[CMakeLists.txt](https://github.com/open62541/open62541/files/11508154/CMakeLists.txt)

Alexk12, 2023-05-18:
It seems you should define the destructor in TEST class where you must call UA_Client_Disconnect and UA_Client_Delete (you do that in connectService(), but does not do on exit from main().

0x00500, 2023-05-19:
> It seems you should define the destructor in TEST class where you must call UA_Client_Disconnect and UA_Client_Delete (you do that in connectService(), but do not do on exit from main().

<< You may not understand what I mean. There is a while loop in the connect Service() function. When my IP is not connected or does not exist, it should try to connect and disconnect in a loop. At this time, on the notebook of the M1 architecture, There will be memory leaks, but no memory leaks on x86 machines.

Alexk12, 2023-05-19:
The loop in connectService() function exits if connection is successful and the m_client_ is set. In the code fragment I do not see how m_client_ is deleted on exit from main() function. (After successful connection).

0x00500, 2023-05-19:
> The loop in connectService() function exits if connection is successful and the m_client_ is set. In the code fragment I do not see how m_client_ is deleted on exit from main() function. (After successful connection).

<< I understand the problem you mentioned, but it is not the cause of the problem. The important thing is that a memory leak occurs when the program enters a loop connection. At this time, the program is running and has not exited the main() function.

Alexk12, 2023-05-19:
If the the program reports memory leaks on the machine where the connection is unsuccessful try to find the exact place where the allocated object is not released. Some debug output is needed where memory leaks are reported.

o6-nbeier, 2026-10-20:
Closing for no activity.

closed 2026-10-20 by o6-nbeier