open62541/open62541 #5791: Support session name in client configuration

Alexk12, 2023-05-21:
## Description

Please support setting sessionName in client configuration for UA_CreateSessionRequest.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake .. -G "Visual Studio 16 2019" -A x64 -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL"
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

OgreTransporter, 2023-06-13:
#5156

RudolfRendier, 2025-08-04:
Looks like this PR for 1.4 implements this:
https://github.com/open62541/open62541/pull/6054/commits/25e51495a1df66e5daec086f719c69efc0635ab2

o6-nbeier, 2026-10-20:
Thank you for opening this issue and for the details you provided.

It describes the same problem as the earlier issue #5156, so we are closing this one as a duplicate to keep the discussion in one place.

closed 2026-10-20 by o6-nbeier