open62541/open62541 #5794: Double free in UA_Array_delete

paddor, 2023-05-23:
## Description

## Background Information / Reproduction Steps

My Ruby binding used to work before I switched to v1.3.6. Previous version was b84c482 on master.
A double free seems to happen while loading a nodeset.

```
==315703== Invalid free() / delete / delete[] / realloc()
==315703==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==315703==    by 0x1160FBA1: UA_Array_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160AF98: String_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160B56A: NodeId_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160DDAC: UA_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF95FF: UA_NodeId_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF9733: deleteRef (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF9792: Node_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF6704: NodeContainer_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF712D: Nodeset_cleanup (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF113F: NodesetLoader_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF009F: NodesetLoader_loadFile (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==  Address 0x29e80bd0 is 0 bytes inside a block of size 9 free'd
==315703==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==315703==    by 0x1160FBA1: UA_Array_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160AF98: String_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160B56A: NodeId_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160DDAC: UA_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF95FF: UA_NodeId_clear (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF9782: Node_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF6704: NodeContainer_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF712D: Nodeset_cleanup (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF113F: NodesetLoader_delete (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF009F: NodesetLoader_loadFile (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11603334: Server_m_load_nodeset (server.c:979)
==315703==  Block was alloc'd at
==315703==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==315703==    by 0x1160F666: UA_Array_copy (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160AF40: String_copy (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x1160DAC6: UA_copy (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AC4A41: UA_String_copy.lto_priv.3 (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AC4EED: parse_nodeid_body (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AC517D: parse_nodeid (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AC5318: UA_NodeId_parse (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF6A22: extractNodedId (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF735F: extractAttributes (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF78B4: initNode (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==315703==    by 0x11AF7907: Nodeset_newNode (in /home/user/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
```

Using git tag v1.3.6 (1b7e2b5d669661c568fda5577ee92011a029cf75).
Using nodesetLoader master @ 4148d22cb41d07f28d38c9517336f6ff0187f474 (Thu Apr 13 23:11:52 2023).
Using Ubuntu Linux 22.04.

Used CMake options:

```
-DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g'
```

Source of `Server_m_load_nodeset()` is:
```
static VALUE Server_m_load_nodeset(VALUE self, VALUE path)
{
  HandlerData *server_data;
  bool       success;

  TypedData_Get_Struct(self, HandlerData, &ServerData_type, server_data); /* unwrap */
  success = NodesetLoader_loadFile(server_data->server, StringValueCStr(path), NULL);

  if (!success)
  {
    rb_raise(rb_eRuntimeError, "nodeset import failed");
  }

  return Qnil;
}
```

matkonnerth, 2023-06-02:
I've add a test with a string nodeId here: https://github.com/open62541/open62541-nodeset-loader/pull/228 but cannot reproduce the memory issue. Can you provide a test nodeset?

paddor, 2023-06-20:
This is a minimal nodeset:
```
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:s1="urn:WS:LTA:Vars/Types.xsd">
  <NamespaceUris>
    <Uri>urn:WS:LTA:Vars</Uri>
  </NamespaceUris>
  <Aliases>
    <Alias Alias="Boolean">i=1</Alias>
    <Alias Alias="SByte">i=2</Alias>
    <Alias Alias="Byte">i=3</Alias>
    <Alias Alias="Int16">i=4</Alias>
    <Alias Alias="UInt16">i=5</Alias>
    <Alias Alias="Int32">i=6</Alias>
    <Alias Alias="UInt32">i=7</Alias>
    <Alias Alias="Int64">i=8</Alias>
    <Alias Alias="UInt64">i=9</Alias>
    <Alias Alias="Float">i=10</Alias>
    <Alias Alias="Double">i=11</Alias>
    <Alias Alias="String">i=12</Alias>
    <Alias Alias="DateTime">i=13</Alias>
    <Alias Alias="ByteString">i=15</Alias>
    <Alias Alias="NodeId">i=17</Alias>
    <Alias Alias="StatusCode">i=19</Alias>
    <Alias Alias="LocalizedText">i=21</Alias>
    <Alias Alias="BaseDataType">i=24</Alias>
    <Alias Alias="Enumeration">i=29</Alias>
    <Alias Alias="Organizes">i=35</Alias>
    <Alias Alias="HasEventSource">i=36</Alias>
    <Alias Alias="HasModellingRule">i=37</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasSubtype">i=45</Alias>
    <Alias Alias="HasProperty">i=46</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
    <Alias Alias="HasNotifier">i=48</Alias>
    <Alias Alias="HasCondition">i=9006</Alias>
    <Alias Alias="BaseObjectType">i=58</Alias>
    <Alias Alias="FolderType">i=61</Alias>
    <Alias Alias="BaseDataVariableType">i=63</Alias>
    <Alias Alias="BaseEventType">i=2041</Alias>
    <Alias Alias="ConditionType">i=2782</Alias>
    <Alias Alias="AcknowledgeableConditionType">i=2881</Alias>
    <Alias Alias="AlarmConditionType">i=2915</Alias>
    <Alias Alias="DiscreteAlarmType">i=10523</Alias>
    <Alias Alias="OffNormalAlarmType">i=10637</Alias>
    <Alias Alias="ConditionVariableType">i=9002</Alias>
    <Alias Alias="TwoStateVariableType">i=8995</Alias>
    <Alias Alias="PropertyType">i=68</Alias>
    <Alias Alias="ModellingRuleType">i=77</Alias>
    <Alias Alias="ObjectsFolder">i=85</Alias>
    <Alias Alias="IdType">i=256</Alias>
    <Alias Alias="NumericRange">i=291</Alias>
    <Alias Alias="UtcTime">i=294</Alias>
    <Alias Alias="Argument">i=296</Alias>
    <Alias Alias="DataItemType">i=2365</Alias>
    <Alias Alias="EnumValueType">i=7594</Alias>
  </Aliases>
  <UAObject NodeId="ns=1;s=CMB_VMS" BrowseName="1:CMB_VMS">
    <DisplayName>CMB_VMS</DisplayName>
    <References>
      <Reference ReferenceType="Organizes" IsForward="false">ObjectsFolder</Reference>
      <Reference ReferenceType="HasComponent">ns=1;s=CMB_VMS.Test</Reference>
      <Reference ReferenceType="HasTypeDefinition">FolderType</Reference>
    </References>
  </UAObject>
  <UAObject NodeId="ns=1;s=CMB_VMS.Test" BrowseName="1:Test" ParentNodeId="ns=1;s=CMB_VMS">
    <DisplayName>Test</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;s=CMB_VMS</Reference>
      <Reference ReferenceType="HasTypeDefinition">BaseObjectType</Reference>
[... 32 lines cut ...]
==817949==    by 0x114BAF98: String_clear (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x114BB56A: NodeId_clear (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x114BDDAC: UA_clear (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A95FF: UA_NodeId_clear (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A9782: Node_delete (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A6704: NodeContainer_delete (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A712D: Nodeset_cleanup (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A113F: NodesetLoader_delete (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A009F: NodesetLoader_loadFile (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x114B3334: Server_m_load_nodeset (server.c:979)
==817949==  Block was alloc'd at
==817949==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==817949==    by 0x114BF666: UA_Array_copy (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x114BAF40: String_copy (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x114BDAC6: UA_copy (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x11974A41: UA_String_copy.lto_priv.3 (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x11974EED: parse_nodeid_body (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x1197517D: parse_nodeid (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
[... cut ...]
==817949==    by 0x119A6A22: extractNodedId (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A735F: extractAttributes (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A78B4: initNode (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==    by 0x119A7907: Nodeset_newNode (in /home/roadster/.gem/ruby/3.2.1/gems/open62541-0.12.4/lib/open62541/open62541.so)
==817949==
```

paddor, 2023-06-22:
More findings: The order of the `UAObject` nodes below seems to matter.
```
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:s1="urn:WS:LTA:Vars/Types.xsd">
  <NamespaceUris>
    <Uri>urn:WS:LTA:Vars</Uri>
  </NamespaceUris>
  <Aliases>
    <Alias Alias="Organizes">i=35</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="ObjectsFolder">i=85</Alias>
    <Alias Alias="FolderType">i=61</Alias>
  </Aliases>

  <!-- works if placed before UAObject "ns=1;s=CMB VMS" -->
  <!--
  <UAObject NodeId="ns=1;s=CMB VMS.Test" BrowseName="1:Test" ParentNodeId="ns=1;s=CMB VMS">
    <DisplayName>Test</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;s=CMB VMS</Reference>
      <Reference ReferenceType="HasTypeDefinition">BaseObjectType</Reference>
    </References>
  </UAObject>
  -->

  <UAObject NodeId="ns=1;s=CMB VMS" BrowseName="1:CMB VMS">
    <DisplayName>CMB VMS</DisplayName>
    <References>
      <Reference ReferenceType="Organizes" IsForward="false">ObjectsFolder</Reference>
      <Reference ReferenceType="HasComponent">ns=1;s=CMB VMS.Test</Reference>
      <Reference ReferenceType="HasTypeDefinition">FolderType</Reference>
    </References>
  </UAObject>

  <!-- crashes if placed after UAObject "ns=1;s=CMB VMS" -->
  <UAObject NodeId="ns=1;s=CMB VMS.Test" BrowseName="1:Test" ParentNodeId="ns=1;s=CMB VMS">
    <DisplayName>Test</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;s=CMB VMS</Reference>
      <Reference ReferenceType="HasTypeDefinition">BaseObjectType</Reference>
    </References>
  </UAObject>

</UANodeSet>
```

And if the Ruby gem is built on Ubuntu 20.04, and then installed on 22.04, it works too. Any ideas?

paddor, 2023-06-29:
The recent merges in nodesetLoader seem to have fixed this issue.
It probably was fixed by one of:
* open62541/nodesetLoader#232
* open62541/nodesetLoader#233
* open62541/nodesetLoader#236

closed 2023-06-29 by paddor