open62541/open62541 #5795: 1.3.6 Doesn't compile on Ubuntu 22.04

paddor, 2023-05-24:
v1.3.6 won't compile anymore on Ubuntu 22.04.

It seems some CC flag `-mcet` doesn't work? Unfortunately the mkmf.log file doesn't say much:

```
$ less tmp/x86_64-linux/open62541/3.2.2/mkmf.log
find_executable: checking for cmake... -------------------- yes

--------------------

find_executable: checking for pkg-config... -------------------- yes

--------------------

find_executable: checking for conan... -------------------- yes

--------------------
```

```
checking for cmake... yes
CMake executable: /usr/bin/cmake
CMake version 3.22 is OK
checking for pkg-config... yes
checking for conan... yes
/usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/home/user/dev/oss/open62541-ruby/tmp/x86_64-linux/open62541/3.2.2/stage -DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF /home/user/dev/oss/open62541-ruby/ext/open62541/deps/open62541
-- open62541 Version: v1.3.6
-- The selected architecture is: posix
-- Test CC flag -std=c99
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -pipe
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wall
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wextra
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wpedantic
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Werror
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-static-in-inline
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-overlength-strings
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-unused-parameter
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wc++-compat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-security
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-nonliteral
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wmissing-prototypes
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wstrict-prototypes
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wredundant-decls
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wuninitialized
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Winit-self
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wcast-qual
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wstrict-overflow
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wnested-externs
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wmultichar
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wundef
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -fno-strict-aliasing
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -fexceptions
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -fstack-protector-strong
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -fstack-clash-protection
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -mcet
-- Performing Test flag_supported
-- Performing Test flag_supported - Failed
-- Test CC flag -fcf-protection
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/dev/oss/open62541-ruby/ext/open62541/deps/open62541
make
make: *** No targets specified and no makefile found.  Stop.
Could not compile open62541
*** ../../../../ext/open62541/extconf.rb failed ***
Could not create Makefile due to some reason, probably lack of necessary
libraries and/or headers.  Check the mkmf.log file for more details.  You may
need configuration options.

Provided configuration options:
        --with-opt-dir
        --without-opt-dir
        --with-opt-include
        --without-opt-include=${opt-dir}/include
        --with-opt-lib
        --without-opt-lib=${opt-dir}/lib
        --with-make-prog
        --without-make-prog
        --srcdir=../../../../ext/open62541
        --curdir
        --ruby=/home/user/.rubies/ruby-3.2.2/bin/$(RUBY_BASE_NAME)
```

Any ideas? No updates have been installed ever since the last time the build worked. Rebooted but the problem persists.

JohannesKauffmann, 2023-05-24:
On WSL2, Ubuntu 22.04, can't reproduce with:
```
git clone https://github.com/open62541/open62541.git
cd open62541
git checkout v1.3.6
git submodule update --init --recursive
mkdir build && cd build
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF ..
make
```

This looks more like a problem of the ruby build system you appear to be using.

paddor, 2023-05-24:
@JohannesKauffmann okay, thanks.

closed 2023-05-24 by paddor