open62541/open62541 #5806: Add multi dimension array to variable value callback

SmallfaceR, 2023-06-05:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):
 - [ x] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x ] Self-contained code example attached
 - [ ] Critical issue

I have created a sample array and tried to assign the dimensions for the Variant. But, I am getting BadTypeMismatch error.

I add the variable to a node. This is working fine. So with UW_expert I can read the variable. The problem comes with the callback routine

When I use the callback function to write new values to the variable I create a array d and Copy it over to a Variant. After this I get a failure bad_mismatch failure. When I set the v.arrayDimensions after the "UA_Variant_setArrayCopy" the OPC server crashes.

Can someone help me to actualize the values in the callback routine?
[image]

[image]
[image]

mlang-de, 2023-06-27:
you already check the variable magtrix exmaple? 
https://github.com/open62541/open62541/blob/fdedfe087299a80f06f8dc089ed59a2721fe6330/examples/tutorial_server_variable.c#L42

mlang-de, 2023-06-27:
within your "beforeRadRF1_array", which imho shall be called "beforeRadRF1_matrix", you write to v.arrayDimensions[0] and [1] without allocating memory before.
"UA_Array_new(4" is wrong, shall be "UA_Array_new(2" since you describe only 2 dimensions.

SmallfaceR, 2023-06-27:
Hello thanks for your help. I changed the callback function but always get this failure. I cant find the failure in the cllback function.
[image]
[image]