open62541/open62541 #5808: VendorServerInfo object node

embbo, 2023-06-07:
## Description
I've been navigating through the server object nodes and found various information. However, when I attempted to explore the 'VendorServerInfo' node, I found that it doesn't have any subnodes. I was anticipating that it would have subnodes similar to other nodes. How can I create subnodes under the 'VendorServerInfo' node, or is it meant to remain empty?
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2023-06-27:
It's up to you whether it is used or not, but "This Object is required to be present even if there are no vendor-defined Objects beneath it".

https://reference.opcfoundation.org/Core/Part5/v104/docs/6.3.1

> VendorServerInfo represents the browse entry point for vendor-defined Server information. This Object is required to be present even if there are no vendor-defined Objects beneath it. See 6.3.6 for its description. 

https://reference.opcfoundation.org/Core/Part5/v104/docs/6.3.6#_Ref135191796

> This ObjectType defines a placeholder Object for vendor-specific information about the OPC UA Server. This ObjectType defines an empty ObjectType that has no components. It shall be subtyped by vendors to define their vendor-specific information.

embbo, 2023-07-04:
@mlang-de 
Are you suggesting that I need to manually create sub-nodes for the vendor information if I want to represent it, Alternatively, how can I add information? Is there a way to include data like in build information "uaServerConfig->buildInfo.productUri "

o6-nbeier, 2026-10-20:
Thank you for the question.

The node is an intentionally empty placeholder, so add your vendor-specific information with the normal information-model APIs (ideally as a VendorServerInfoType subtype).

Closing for now.

closed 2026-10-20 by o6-nbeier