open62541/open62541 #5811: Error in CMakeLists.txt file - ua_pubsub_mqtt.c and other files are excluded from build on Windows platform

Alexk12, 2023-06-08:
Error in CMakeLists.txt file - ua_pubsub_mqtt.c and other files are excluded from build on Windows platform

CMAKE options:
cmake .. -G "Visual Studio 16 2019" -A x64 -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL" -DUA_ENABLE_MQTT=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON

From open62541\CMakeLists.txt

Was (with error in if/else logic):
......
    if(UA_ENABLE_PUBSUB_MQTT)
        if(WIN32)
            MESSAGE(WARNING "Multithreading is enabled in MQTT plugin. This feature is under development and marked as EXPERIMENTAL")
        else()
            list(APPEND default_plugin_headers ${PROJECT_SOURCE_DIR}/deps/mqtt-c/include/mqtt_pal.h
                 ${PROJECT_SOURCE_DIR}/deps/mqtt-c/include/mqtt.h
                 ${PROJECT_SOURCE_DIR}/plugins/include/open62541/plugin/pubsub_mqtt.h
                 ${PROJECT_SOURCE_DIR}/plugins/mqtt/ua_mqtt-c_adapter.h)
            list(APPEND default_plugin_sources ${PROJECT_SOURCE_DIR}/deps/mqtt-c/src/mqtt_pal.c
                 ${PROJECT_SOURCE_DIR}/deps/mqtt-c/src/mqtt.c
                 ${PROJECT_SOURCE_DIR}/plugins/ua_pubsub_mqtt.c
                 ${PROJECT_SOURCE_DIR}/plugins/mqtt/ua_mqtt-c_adapter.c)
        endif()
    endif()
......

Correct:
......
    if(UA_ENABLE_PUBSUB_MQTT)
        if(WIN32)
            MESSAGE(WARNING "Multithreading is enabled in MQTT plugin. This feature is under development and marked as EXPERIMENTAL")
        endif()
        list(APPEND default_plugin_headers ${PROJECT_SOURCE_DIR}/deps/mqtt-c/include/mqtt_pal.h
             ${PROJECT_SOURCE_DIR}/deps/mqtt-c/include/mqtt.h
             ${PROJECT_SOURCE_DIR}/plugins/include/open62541/plugin/pubsub_mqtt.h
             ${PROJECT_SOURCE_DIR}/plugins/mqtt/ua_mqtt-c_adapter.h)
        list(APPEND default_plugin_sources ${PROJECT_SOURCE_DIR}/deps/mqtt-c/src/mqtt_pal.c
             ${PROJECT_SOURCE_DIR}/deps/mqtt-c/src/mqtt.c
             ${PROJECT_SOURCE_DIR}/plugins/ua_pubsub_mqtt.c
             ${PROJECT_SOURCE_DIR}/plugins/mqtt/ua_mqtt-c_adapter.c)
    endif()
......

## Description
Error in CMakeLists.txt file - ua_pubsub_mqtt.c and other files are excluded from build on Windows platform

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake .. -G "Visual Studio 16 2019" -A x64 -DUA_ENABLE_ENCRYPTION:STRING="OPENSSL" -DUA_ENABLE_MQTT=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-07-14:
Try the 1.4 branch for the next release.
That code was completely refactored.

Alexk12, 2023-07-14:
> Try the 1.4 branch for the next release. That code was completely refactored.

Thank you, I'll try new release.

o6-nbeier, 2026-10-22:
Closing for no activity.

closed 2026-10-22 by o6-nbeier