open62541/open62541 #5814: CS PADIM could not be compiled

ccvca, 2023-06-12:
## Description

Using the code generator for the PADIM-Specification, the following code is generated:

```c++
#define UA_TYPES_PADIM_COUNT 0
#define UA_TYPES_PADIM NULL
```

```c++
/* Change namespaceIndex from current namespace */
for(int i = 0; i < UA_TYPES_PADIM_COUNT; i++) {
UA_TYPES_PADIM[i].typeId.namespaceIndex = ns[2];
UA_TYPES_PADIM[i].binaryEncodingId.namespaceIndex = ns[2];
}
```
As UA_TYPES_PADIM is "void* NULL", this results in the follwing error:
```c++
/[...]/src_generated/namespace_padim_generated.c:4809:15: warning: dereferencing ‘void *’ pointer
 4809 | UA_TYPES_PADIM[i].typeId.namespaceIndex = ns[2];
      |               ^
/[...]/src_generated/namespace_padim_generated.c:4809:18: error: request for member ‘typeId’ in something not a structure or union
 4809 | UA_TYPES_PADIM[i].typeId.namespaceIndex = ns[2];
      |                  ^
/[...]/src_generated/namespace_padim_generated.c:4810:15: warning: dereferencing ‘void *’ pointer
 4810 | UA_TYPES_PADIM[i].binaryEncodingId.namespaceIndex = ns[2];
      |               ^
/[...]/src_generated/namespace_padim_generated.c:4810:18: error: request for member ‘binaryEncodingId’ in something not a structure or union
 4810 | UA_TYPES_PADIM[i].binaryEncodingId.namespaceIndex = ns[2];
      |                  ^

```

The code is generated by these lines:

https://github.com/open62541/open62541/blob/f481959ca3330fb24f81e23383a5e007f68e9fe3/tools/nodeset_compiler/backend_open62541.py#L311-L319

This condition is not sufficiant: `if len(typesArray) > 0:` But I don't known how to get the real length of the types.

 - [x] open62541 Version (release number or git tag): master

GoetzGoerisch, 2023-06-12:
@ccvca thank you for the report.

@NoelGraf could you please look into this?

NoelGraf, 2023-06-12:
@ccvca can you provide your CMake file in which the PADIM specification is generated?

However, I already suspect what the problem is. PADIM defines its own data types but does not provide a bsd file. Also the xml file does not contain a base64 decoded block from which the bsd file can be generated.
The basic problem is that the nodeset compiler needs the bsd even though it is deprecated.
With the nodesetinjector changes have been made. Thus, for example, the specification of the NamespaceMap is no longer necessary. Now it is checked at runtime which index the loaded nodeset has and is adapted accordingly in the types array.
For this it is checked whether the nodeset defines own data types and thus the Types Array must be adapted.

GoetzGoerisch, 2023-06-12:
@NoelGraf Yes, Companion specification no longer include base64 encoding nodes for the custom datatypes due to being deprecated. Therefore tools like UaModeler no longer include encoding nodes.

The DataTypeDefinition Attribute concept shall be followed:
https://reference.opcfoundation.org/Core/Part5/v105/docs/B

CMake reference; https://github.com/umati/Sample-Server/blob/d25ce13a1e539a534a60ee01b56c4ca59e411f04/CMakeLists.txt#L264

ccvca, 2023-06-13:
As there is no bsd-File provided, an empty one is used to use the cmake-functions.

PS: @NoelGraf The current solution of these namespace adaptations might cause problems in future applications, see #5813

GoetzGoerisch, 2023-06-26:
@NoelGraf how should we move forward with this?

NoelGraf, 2023-06-28:
@GoetzGoerisch For now, I would tweak it so that for namespaces without a bsd file, an empty bsd is automatically generated. The code that modifies the types array and sets the correct index is only included if the types array contains any data at all. This way, PADIM can be built and other nodesets can be integrated without a bsd file and without a base64 block. 
The general problem with the required BSD file needs to be addressed in the near future, as more and more nodesets do not provide a bsd file (e.g. Safety, Glass).
However, this will require some re-working of the nodeset compiler.

I will revisit issue #5813 in the next few days and come up with a solution.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The empty datatype namespace is guarded now and PADIM is integrated in #5851, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier