open62541/open62541 #5815: Help RuntimeError: Infinite loop detected or type not found while processing type?

farhan296, 2023-06-12:
## Description
Hello,

I need help with the following. I am working on an application that makes use of [Safety Nodeset](https://github.com/OPCFoundation/UA-Nodeset/tree/d9f49576eedb95f4ffaa731bad09779947b287f5/Safety). But I run into an error while compiling the nodesets. Can anyone help me with this. The details are as follows.

## Background Information / Reproduction Steps
1. I generate Nodeset2 and related files using UA-Model Compiler.

For Safety Nodeset: 
```bash
Opc.Ua.ModelCompiler.exe compile-nodesets -input .\OPC_Foundation_Safety-files -o2 .\generated_safety -uri http://opcfoundation.org/UA/Safety

copy .\generated_safety .\generated
```

For App.xml
```bash
del .\AppModel.csv

Opc.Ua.ModelCompiler.exe compile -d2 .\AppModel.xml -d2 .\OPC_Foundation_Safety-files\Opc.Ua.Safety.NodeSet2.xml,Opc.Ua.Safety,Safety -cg .\AppModel.csv -version v105 -o2 .\generated
```
And the AppModel.xml is as follows
```XML
<?xml version="1.0" encoding="utf-8"?>

<ModelDesign
  xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:ua="http://opcfoundation.org/UA/"
  xmlns:Safety="http://opcfoundation.org/UA/Safety"
  xmlns:app="urn:open62541.server.application"
  xmlns:xsd="http://www.w3.org/2001/XMLSchema"
  TargetNamespace="urn:open62541.server.application"
  TargetVersion="0.1.0"
  TargetPublicationDate="2022-05-02T00:00:00Z"
  xmlns="http://opcfoundation.org/UA/ModelDesign.xsd">

  <Namespaces>
    <Namespace Name="app" Prefix="app">urn:open62541.server.application</Namespace>
    <Namespace Name="Safety" Prefix="Safety" PublicationDate="2022-06-28T00:00:00Z" Version="1.05.02">http://opcfoundation.org/UA/Safety</Namespace>
    <Namespace Name="ua" Prefix="ua" Version="1.05.02" PublicationDate="2022-11-01T00:00:00Z">http://opcfoundation.org/UA/</Namespace>
  </Namespaces>

  <DataType SymbolicName="app:MyResponseSPDUDataType" BaseType="Safety:ResponseSPDUDataType" IsAbstract="false">
    <Fields>      
      <Field Name="MyOutSafetyData"  DataType="app:MySafetyData1Type"></Field>
      <Field Name="MyOutNonSafetyData" DataType="app:MyNonSafetyData1Type"></Field>
    </Fields>
  </DataType>

  <DataType SymbolicName="app:MySafetyData1Type" BaseType="ua:Structure" >   
    <Fields>
      <Field Name="Field_1" DataType="ua:UInt32"></Field>
      <Field Name="Field_2" DataType="ua:UInt32"></Field>
      <Field Name="Field_4" DataType="ua:Boolean"></Field>     
    </Fields>
  </DataType>

  <DataType SymbolicName="app:MyNonSafetyData1Type" BaseType="ua:Structure">
    <Fields>
      <Field Name="Item1" DataType="ua:UInt32"></Field>
    </Fields>
  </DataType>

  <!-- ### Object ###-->

  <Object SymbolicName="app:MySafetyProvider" TypeDefinition="Safety:SafetyProviderType">
    <Children>
      <Method SymbolicName="app:MyReadSafetyData">
        <InputArguments>
          <Argument Name="InSafetyConsumerID" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="InMonitoringNumber" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="InFlags" DataType="Safety:InFlagsType" ValueRank="Scalar"></Argument>
        </InputArguments>
        <OutputArguments>
          <Argument Name="OutSafetyData" DataType="app:MySafetyData1Type" ></Argument>
          <Argument Name="OutFlags" DataType="Safety:OutFlagsType" ValueRank="Scalar"></Argument>
          <Argument Name="OutSPDU_ID_1" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutSPDU_ID_2" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutSPDU_ID_3" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutSafetyConsumerID" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutMonitoringNumber" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutCRC" DataType="ua:UInt32" ValueRank="Scalar"></Argument>
          <Argument Name="OutNonSafetyData" DataType="app:MyNonSafetyData1Type" ></Argument>
        </OutputArguments>
      </Method>           
      <Object SymbolicName="app:MySafetyPDUs" TypeDefinition="Safety:SafetyPDUsType">
        <Children>
          <Variable SymbolicName="app:MyRequestSPDU" DataType="Safety:RequestSPDUDataType" TypeDefinition="ua:BaseDataVariableType"></Variable>
          <Variable SymbolicName="app:MyResponseSPDU" DataType="app:MyResponseSPDUDataType" TypeDefinition="ua:BaseDataVariableType"></Variable>
        </Children>
      </Object>
    </Children>

    <References>
      <Reference IsInverse="true">
        <ReferenceType>ua:Organizes</ReferenceType>
        <TargetId>Safety:SafetyACSet</TargetId>
      </Reference>
    </References>
  </Object>   
  
</ModelDesign>

```

Then I used the CMake GUI to generate Visual Studio Project. The generator is set to Visual Studio 17 2022
```CMake
cmake_minimum_required(VERSION 3.2)
project(app VERSION 0.1.0)

find_package(open62541 REQUIRED COMPONENTS Events FullNamespace)

ua_generate_nodeset_and_datatypes(
    NAME "safety"
    FILE_NS "${PROJECT_SOURCE_DIR}/generated/UAModel.Safety.NodeSet2.xml"
    FILE_CSV "${PROJECT_SOURCE_DIR}/generated/UAModel.Safety.NodeIds.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/generated/UAModel.Safety.Types.bsd"
    NAMESPACE_MAP "2:http://opcfoundation.org/UA/Safety"
    INTERNAL
)

ua_generate_nodeset_and_datatypes(
    NAME "app"
    FILE_NS "${PROJECT_SOURCE_DIR}/generated/app.NodeSet2.xml"
    FILE_CSV "${PROJECT_SOURCE_DIR}/generated/app.NodeIds.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/generated/app.Types.bsd" "${PROJECT_SOURCE_DIR}/generated/UAModel.Safety.Types.bsd"
    NAMESPACE_MAP "1:urn:open62541.server.application"
    DEPENDS "safety"
    INTERNAL
    )
	
add_executable(${PROJECT_NAME} main.c)
target_link_libraries(${PROJECT_NAME} open62541::open62541)
```
Used CMake options:
```bash
-DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release;MinSizeRel;RelWithDebInfo" -Dopen62541_DIR:PATH="./../../../SDK_Open62541/open62541/build" -DUA_NAMESPACE_ZERO:STRING="FULL" -DCMAKE_INSTALL_PREFIX:PATH="C:/Program Files (x86)/app" -Dopen62541_TOOLS_DIR:PATH="./../../../SDK_Open62541/open62541/tools" -DUA_NODESET_DIR:PATH="./../../../SDK_Open62541/open62541/deps/ua-nodeset"
```

When I compile on Visual Studio 2022 I get the following error:

```bash
5>Traceback (most recent call last):
5>  File ".\..\..\..\SDK_Open62541\open62541\tools\generate_datatypes.py", line 95, in <module>
5>    parser.create_types()
5>  File ".\..\..\..\SDK_Open62541\open62541\tools\nodeset_compiler\type_parser.py", line 377, in create_types
5>    self.parse_types()
5>  File ".\..\..\..\SDK_Open62541\open62541\tools\nodeset_compiler\type_parser.py", line 419, in parse_types
5>    self.parseTypeDefinitions(self.outname, f)
5>  File ".\..\..\..\SDK_Open62541\open62541\tools\nodeset_compiler\type_parser.py", line 321, in parseTypeDefinitions
5>    raise RuntimeError("Infinite loop detected or type not found while processing types " +
5>RuntimeError: Infinite loop detected or type not found while processing types MyResponseSPDUDataType: unknonwn subtype ['Opc.Ua.Safety:OutFlagsType']. If the unknown subtype is 'Bit', then maybe a struct with optional fields is defined wrong in the .bsd-file. If not, maybe you need to import additional types with the --import flag. E.g. '--import=UA_TYPES#/path/to/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd'

```

Can anyone help on how to resolve the error with ['Opc.Ua.Safety:OutFlagsType'] Thanks,

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): 90271982a0ca278bd9887c5ca131ebe06c962d83
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

farhan296, 2023-06-15:
@jpfr any insights into this will be much appreciated. Thanks

farhan296, 2023-06-15:
It seems the issue is arising at [line 431 of nodeset.py](https://github.com/open62541/open62541/blob/0b1d72f157c6a8f0c05aca56896b217c816e9827/tools/nodeset_compiler/nodeset.py#L431) file. It expects a node with name *_BinarySchema or TypeDictionary_BinarySchema which in case of Safety companion nodeset does not exist. Due to this the UAModel.Safety.Types.bsd file is not included as it does not contain any node with the name _BinarySchema or  TypeDictionary_BinarySchema. The UA Model compiler does not generate above nodes. Is there any method to make this work?

NoelGraf, 2023-07-04:
@farhan296 
The nodeset compiler needs the bsd file for the nodeset. If this is not available, it looks in the xml file for the base64 decoded block. The bsd file is extracted from this. Safety does not provide a bsd file in the current version and does not include the base64 block in the xml file.
In the Cmake you specify a bsd file for Saftey. Is this the correct bsd file?

NoelGraf, 2023-07-25:
Close the issue for now, reopen it if necessary.

closed 2023-07-25 by NoelGraf