open62541/open62541 #5819: Bug in UA_Array_copy

happybruce, 2023-06-15:
## Description
Recently i got seg fault when monitor variable change at server side. After debugging, find that root cause happen in UA_Array_copy, the arg "src" is NULL, "size" is 1, then it will call memcpy, this cause seg fault. **The code do not check validation of "src" before copy  mem.**  

```
UA_StatusCode
UA_Array_copy(const void *src, size_t size,
              void **dst, const UA_DataType *type) {
    if(size == 0) {
        if(src == NULL)
            *dst = NULL;
        else
            *dst= UA_EMPTY_ARRAY_SENTINEL;
        return UA_STATUSCODE_GOOD;
    }

    if(!type)
        return UA_STATUSCODE_BADINTERNALERROR;

    /* calloc, so we don't have to check retval in every iteration of copying */
    *dst = UA_calloc(size, type->memSize);
    if(!*dst)
        return UA_STATUSCODE_BADOUTOFMEMORY;

    if(type->pointerFree) {
        memcpy(*dst, src, type->memSize * size);
        return UA_STATUSCODE_GOOD;
    }

    uintptr_t ptrs = (uintptr_t)src;
    uintptr_t ptrd = (uintptr_t)*dst;
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    for(size_t i = 0; i < size; ++i) {
        retval |= UA_copy((void*)ptrs, (void*)ptrd, type);
        ptrs += type->memSize;
        ptrd += type->memSize;
    }
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Array_delete(*dst, size, type);
        *dst = NULL;
    }
    return retval;
}
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version 1.3.6

happybruce, 2023-06-15:
Currently i don‘t know why src is NULL

jpfr, 2023-08-03:
Can you share a backtrace of the bug?
The bug happens *before* copy is called.

nimarty, 2023-11-13:
I'm quite sure we're observing the same issue in our project. Here's a backtrace:
```
Core was generated by `/opt/myorg/app/MyApp'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:109
109     ../sysdeps/aarch64/multiarch/../memcpy.S: No such file or directory.
[Current thread is 1 (LWP 79535)]
(gdb) bt
#0  __memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:109
#1  0x0000aaaad2478590 in memcpy (__len=<optimized out>, __src=0x0, __dest=<optimized out>)
    at /opt/mydistro/3.3/sysroots/cortexa53-crypto-poky-linux/usr/include/bits/string_fortified.h:29
#2  UA_Array_copy (src=0x0, size=1, dst=0xffff611604a0, type=0xaaaad31a7f20 <UA_TYPES+360>)
    at ../../../../../libs/my_project/libs/open62541/src/ua_types.c:1715
#3  0x0000aaaad24787e0 in UA_Array_copy (type=<optimized out>, dst=<optimized out>, size=<optimized out>,
    src=<optimized out>) at ../../../../../libs/my_project/libs/open62541/src/ua_types.c:1706
#4  Variant_copy (_=0x0, dst=0xffff61160488, src=0xffff7038bc18)
    at ../../../../../libs/my_project/libs/open62541/src/ua_types.c:553
#5  DataValue_copy (src=0xffff7038bc18, dst=0xffff61160488, _=<optimized out>)
    at ../../../../../libs/my_project/libs/open62541/src/ua_types.c:944
#6  0x0000aaaad24810a8 in UA_copy (type=<optimized out>, dst=0xffff61160488, src=0xffff7038bc18)
    at ../../../../../libs/my_project/libs/open62541/src/ua_types.c:1153
#7  UA_DataValue_copy (dst=0xffff61160488, src=0xffff7038bc18) at src_generated/open62541/types_generated_handling.h:714
#8  readValueAttributeFromNode (server=0xffff7026e170, session=<optimized out>, vn=0xffff7038bad0, v=0xffff61160488,
    rangeptr=0x0) at ../../../../../libs/my_project/libs/open62541/src/server/ua_services_attribute.c:143
#9  0x0000aaaad2481320 in readValueAttributeComplete (server=server@entry=0xffff7026e170,
    session=session@entry=0xffff7026e4d0, vn=vn@entry=0xffff7038bad0,
    timestamps=timestamps@entry=UA_TIMESTAMPSTORETURN_SOURCE, indexRange=indexRange@entry=0xffff7038b990,
    v=v@entry=0xffff61160488) at ../../../../../libs/my_project/libs/open62541/src/server/ua_services_attribute.c:227
#10 0x0000aaaad24841f8 in ReadWithNode (node=0xffff7038bad0, server=0xffff7026e170, session=0xffff7026e4d0,
    timestampsToReturn=UA_TIMESTAMPSTORETURN_SOURCE, id=0xffff7038b970, v=0xffff61160488)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_services_attribute.c:446
#11 0x0000aaaad267691c in UA_Server_readWithSession (timestampsToReturn=UA_TIMESTAMPSTORETURN_SOURCE,
    item=0xffff7038b970, session=0xffff7026e4d0, server=0xffff7026e170)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_services_attribute.c:633
#12 monitoredItem_sampleCallback (monitoredItem=0xffff7038b8e0, server=0xffff7026e170)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_subscription_datachange.c:224
--Type <RET> for more, q to quit, c to continue without paging--
#13 UA_MonitoredItem_sampleCallback (server=0xffff7026e170, monitoredItem=0xffff7038b8e0)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_subscription_datachange.c:204
#14 0x0000aaaad268934c in serverExecuteRepeatedCallback (data=<optimized out>, callbackApplication=<optimized out>,
    cb=<optimized out>, server=0xffff7026e170) at ../../../../../libs/my_project/libs/open62541/src/server/ua_server.c:673
#15 UA_Timer_process.constprop.0 (t=t@entry=0xffff7026e648, nowMonotonic=nowMonotonic@entry=3251911772568,
    executionApplication=executionApplication@entry=0xffff7026e170, executionCallback=<optimized out>)
    at ../../../../../libs/my_project/libs/open62541/src/ua_timer.c:263
#16 0x0000aaaad247b5b4 in UA_Server_run_iterate (waitInternal=true, server=0xffff7026e170)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_server.c:680
#17 UA_Server_run (running=<optimized out>, server=<optimized out>)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_server.c:775
#18 UA_Server_run (server=0xffff7026e170, running=0xffff702238a0)
    at ../../../../../libs/my_project/libs/open62541/src/server/ua_server.c:760
#19 0x0000aaaad2456f64 in my_project::opcua::server::OPCUAServer::handleNetworkReady() ()
#20 0x0000ffff908daa5c in std::execute_native_thread_routine (__p=0xffff7063f030)
    at ../../../../../../../../../../work-shared/gcc-10.2.0-r0/gcc-10.2.0/libstdc++-v3/src/c++11/thread.cc:80
#21 0x0000ffff907c4d44 in start_thread (arg=0xffff7b30327f) at pthread_create.c:473
#22 0x0000ffff90718e5c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:77
```

markaluza, 2023-11-15:
I can see same issue... v 1.3.6 compiled with msvc

src = NULL

    if(type->pointerFree) 
    {
        memcpy(*dst, src, type->memSize * size);
        return UA_STATUSCODE_GOOD;
    }

memcpy(unsigned char * dst, unsigned char * src, unsigned long count)
UA_Array_copy(const void * src, unsigned int size, void * * dst, const UA_DataType * type)
[Inline Frame] Variant_copy(const UA_Variant *) 
DataValue_copy(const UA_DataValue * src, UA_DataValue * dst, const UA_DataType * _) 
readValueAttributeComplete(UA_Server * server, UA_Session * session, const UA_VariableNode * vn, UA_TimestampsToReturn timestamps, const UA_String * indexRange, UA_DataValue * v) 
ReadWithNode(const UA_Node * node, UA_Server * server, UA_Session * session, UA_TimestampsToReturn timestampsToReturn, const UA_ReadValueId * id, UA_DataValue * v) 
UA_Server_readWithSession(UA_Server * server, UA_Session * session, const UA_ReadValueId * item, UA_TimestampsToReturn timestampsToReturn)
monitoredItem_sampleCallback(UA_Server * server, UA_MonitoredItem * monitoredItem) 
serverExecuteRepeatedCallback(UA_Server * server, void(*)(void *, void *) cb, void * callbackApplication, void * data) 
UA_Timer_process(UA_Timer * t, __int64 nowMonotonic, void(*)(void *, void(*)(void *, void *), void *, void *) executionCallback, void * executionApplication) 
UA_Server_run_iterate(UA_Server * server, bool waitInternal)

happybruce, 2023-12-05:
So we need solve this bug.

jpfr, 2023-12-05:
There is a solution in https://github.com/open62541/open62541/pull/6161.
This ensures that the application will not crash if we copy an array of length > 0 and a NULL data-pointer.

But this is only a band-aid.
An array of length > 0 and a NULL data-pointer, that should never happen.

Ideally you debug into this and figure out which node contains this value.
From the NodeId it should be possible to find where this value originates from.
It must have been created locally. Because we do not create such inconsistent data when receiving messages over the network.
That part has been debugged and fuzz-tested very thoroughly.

happybruce, 2023-12-06:
Thanks for your reply. 
I see the change, that's OK.  My trace is similar with [nimarty](https://github.com/nimarty), https://github.com/open62541/open62541/issues/5819#issuecomment-1807825820

Normally I am using open62541.c/h, when seg fault happen,  it is very slow to load using gdb...

closed 2023-12-06 by happybruce

cumtyxf, 2024-09-08:
I encountered the same error, the version is 1.3.6, and src is null.
add this code

if(src == NULL)
    return UA_STATUSCODE_BADOUTOFMEMORY;
    
Although the  src  is maybe still occurs null , but the program can  execute normally, no segment fault pop.

jpfr, 2024-09-09:
You should only experience a memfault if src == NULL and size > 0.
That should never happen on your end.
We can add a test for that and return UA_STATUSCODE_BADINTERNALERROR.

nimarty, 2024-09-09:
We added the following build configuration. This solved our problem. I think a fix like `if(src == null) ...` does not solve the root cause of this issue. In our case it was related to multi-threading.

set(UA_ENABLE_IMMUTABLE_NODES ON)
set(UA_MULTITHREADING 100)

cumtyxf, 2024-09-09:
> You should only experience a memfault if src == NULL and size > 0. That should never happen on your end. We can add a test for that and return UA_STATUSCODE_BADINTERNALERROR.

i think size no need to judge.
src == NULL  has already been indicated an error.

cumtyxf, 2024-09-09:
> We added the following build configuration. This solved our problem. I think a fix like `if(src == null) ...` does not solve the root cause of this issue. In our case it was related to multi-threading.
> 
> set(UA_ENABLE_IMMUTABLE_NODES ON) set(UA_MULTITHREADING 100)

I agree with what you said.

PetrKocian, 2024-09-30:
> Ideally you debug into this and figure out which node contains this value.
> From the NodeId it should be possible to find where this value originates from.
> It must have been created locally. Because [...]

I'm running into the same issue. What do you mean by inconsistent data in the node? The value that the node holds? I tried to follow the tutorials when creating the nodes and then writing to them, but I could've introduced some errors when editing the code. Here is my code for adding variables to the server and for writing to them:

```
static void addIntVariable(char* description, char* displayName, char* nodeID, char* nodeName, UA_Int32 defaultValue, UA_NodeId parentNodeId, bool writeable) {
	/* Define the attribute of the variable node */
	UA_VariableAttributes attr = UA_VariableAttributes_default;
	UA_Int32 myInteger = defaultValue;
	UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	attr.description = UA_LOCALIZEDTEXT("en-US", description);
	attr.displayName = UA_LOCALIZEDTEXT("en-US", displayName);
	attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	attr.accessLevel = UA_ACCESSLEVELMASK_READ | (writeable << 1u);
	attr.valueRank = UA_VALUERANK_SCALAR;

	/* Add the variable node to the information model */
	UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, nodeID);
	UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, nodeName);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
		parentReferenceNodeId, myIntegerName,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}
```
```
static void writeIntVariable(char* nodeID, UA_Int32 input) {
	UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, nodeID);

	UA_Int32 myInteger = input;
	UA_Variant myVar;
	UA_Variant_init(&myVar);
	UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	UA_Server_writeValue(server, myIntegerNodeId, myVar);
}
```

I can not see anything that could be causing inconsistent data. Could maybe the parent node be the issue?