open62541/open62541 #5820: Authenticate Username and Password requirements from external programs.

yirongliang, 2023-06-15:
## Description
I want to be able to verify Username and Password myself. Similar to UA_CertificateVerification calling the verify method. It is up to us to check for correctness.
Because my Username and Password may change dynamically during execution, so when authentication is required, I can dynamically query the password. can this be done?

neojaw, 2023-06-15:
The best (and intended) way to do your own user management is to create your own AccessControl plugin and handle the sessions yourself. Take a look at `plugins/ua_accesscontrol_default.c`. You need to implement at least your version of the `activateSession()` (where you use your own mechanism of querying user names and passwords) and `closeSession()` functions and then register them with a function like `UA_AccessControl_default()`. Use the `context` field to pass your own object(s) to the callback functions.

It _might_ also work to acquire and modify the `usernamePasswordLogin` list on the fly, but I don't think that is a good idea.

yirongliang, 2023-06-16:
Thank you for your reply. I have understood how to solve this problem.

closed 2023-06-16 by yirongliang