open62541/open62541 #5829: [1.3] Can not run simple server demo in windows10 with vs2022

CHN-STUDENT, 2023-06-20:
## Description

I use cmake3.26.4 (with-gui) to generating vs solution, then open it in windows10 with vs2022, and create a open62541.lib, but i follow the doc how can[ building a simple server](https://www.open62541.org/doc/1.3/tutorial_server_firststeps.html), i get some error. Could you give me some advice?
I also add `include` folder, `plugins\include` folder and `open62541.lib`  to my vs project.
 
[image]
[image]

CMake Ouput:
```
Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19045.
The C compiler identification is MSVC 19.36.32532.0
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working C compiler: D:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - skipped
Detecting C compile features
Detecting C compile features - done
Found Python3: D:/Python311/python.exe (found version "3.11.4") found components:  Interpreter 
Could NOT find Git (missing: GIT_EXECUTABLE) 
Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at tools/cmake/SetGitBasedVersion.cmake:23 (message):
  Failed to determine the version from git information.  Using defaults.
Call Stack (most recent call first):
  CMakeLists.txt:51 (set_open62541_version)

open62541 Version: v1.2.2-unknown
CMAKE_BUILD_TYPE not given; setting to 'Debug'
The selected architecture is: win32
Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
Could NOT find LATEX (missing: LATEX_COMPILER) 
Configuring done (7.2s)
Generating done (0.3s)
```

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

r-sin, 2023-06-23:
Try enabling UA_BUILD_EXAMPLES in CMake then configure/generate. In VS select one of the examples/tutorials als Startup Project and press F5. At least this should work, or there is realy a problem.

CHN-STUDENT, 2023-06-25:
@r-sin Thanks for your advice, but it sames not work me, i get some error too.

closed 2026-10-20 by o6-nbeier