open62541/open62541 #5831: Connect GDS to a open62541 server

bsallesUFCG, 2023-06-20:
## Description
I'm trying to connect UaGDS from Unified Automation to the LDS-ME example of open62541 (I added security). 
GDS is pushing a certificate but the open62541 server is not accepting this certificate. 
Has anyone tried to connect UaGDS to any open62541 server?
Could you give me some advice?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

masteralarm, 2023-06-30:
To be able to get a new application certificate pushed, the open62541 application needs to support the GDS push model. I don't think that's the intended purpose of the LDS-ME example.

Currently, certificate push management is not supported by open62541, but there is a WIP pull request adding this functionality: [https://github.com/open62541/open62541/pull/5524](https://github.com/open62541/open62541/pull/5524)

erossignon, 2026-08-27:
the https://github.com/open62541/open62541/pull/5524 is now closed so it's in.