open62541/open62541 #5841: Unable to run ”tutorial_pubsub_mqtt_publish.c“ correctly

uniquecash, 2023-06-27:
## Description
When I try to run ”tutorial_pubsub_mqtt_publish.c“ and ”tutorial_pubsub_mqtt_subscribe.c“，there's erro reporting
"[2023-06-27 11:01:36.545 (UTC+0800)] error/pubsub       Connection ns=1;i=52040 | The requested protocol is not supported
[2023-06-27 11:01:36.545 (UTC+0800)] info/userland      Connection destructor called!
[2023-06-27 11:01:36.546 (UTC+0800)] error/server       Error Name = BadNotFound"
what should I do to solve this problem?
## Background Information / Reproduction Steps
[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

xcxcxcx2019, 2023-06-28:
I have the same problem today, and it has not been solved yet.

uniquecash, 2023-06-28:
> I have the same problem today, and it has not been solved yet.

how do you deal with that now?maybe we can communicate about this problem

jpfr, 2023-06-30:
You need to enable MQTT support in cmake.
There are also gui tools like ccmake to configure.

https://github.com/open62541/open62541/blob/ceebefe850a9954cf997055bdc5172c459079ebb/CMakeLists.txt#L446

NoelGraf, 2023-07-04:
@uniquecash Can the issue be closed?

tukotech, 2023-08-22:
I am seeing the same error using master or 1.4 branch. Below is my cmake options:

```
cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DUA_ENABLE_PUBSUB=ON -DENABLE_EXAMPLES=ON -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_MQTT=ON -DUA_FORCE_WERROR=OFF -DUA_ENABLE_PUBSUB_MQTT=ON -DUA_ENABLE_MQTT=ON ..
```

jpfr, 2023-08-22:
You don't need `UA_ENABLE_PUBSUB_MQTT` any longer.
It is now `UA_ENABLE_MQTT`.
In the future we will also have `UA_ENABLE_HTTP`, and so on.
Not just for PubSub.

tukotech, 2023-08-22:
Did make clean and re-run cmake and make, I am still not able to run mqtt. 

cmake command below
```
cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DUA_ENABLE_PUBSUB=ON -DENABLE_EXAMPLES=ON -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_FORCE_WERROR=OFF -DUA_ENABLE_MQTT=ON ..

make -j$(nproc)
```

Output when running binary:
```
user1@mac1:~/prj/open62541/build$ bin/examples/tutorial_pubsub_mqtt_publish
[2023-08-22 11:29:38.872 (UTC-0500)] info/eventloop     Starting the EventLoop
[2023-08-22 11:29:38.872 (UTC-0500)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-08-22 11:29:38.872 (UTC-0500)] info/server        AccessControl: Anonymous login is enabled
[2023-08-22 11:29:38.872 (UTC-0500)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2023-08-22 11:29:38.874 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2023-08-22 11:29:38.879 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=25451): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2023-08-22 11:29:38.893 (UTC-0500)] warn/server        Writing the value of Node i=12779 failed with the following reason: DataType of the value is incompatible
[2023-08-22 11:29:38.893 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | WriteRequest returned status code BadTypeMismatch
[2023-08-22 11:29:38.893 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=12779): Could not create a default value with StatusCode BadTypeMismatch
[2023-08-22 11:29:38.893 (UTC-0500)] warn/server        Writing the value of Node i=2165 failed with the following reason: DataType of the value is incompatible
[2023-08-22 11:29:38.893 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | WriteRequest returned status code BadTypeMismatch
[2023-08-22 11:29:38.893 (UTC-0500)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (i=2165): Could not create a default value with StatusCode BadTypeMismatch
[2023-08-22 11:29:38.903 (UTC-0500)] error/pubsub       Connection ns=1;i=52040 | The requested protocol is not supported
[2023-08-22 11:29:38.903 (UTC-0500)] info/userland      Connection destructor called!
[2023-08-22 11:29:38.903 (UTC-0500)] error/server       Error Name = BadNotFound
[2023-08-22 11:29:38.903 (UTC-0500)] info/server        PubSub cleanup was called.
[2023-08-22 11:29:38.903 (UTC-0500)] info/userland      PublishedDataItems destructor called!
[2023-08-22 11:29:38.903 (UTC-0500)] info/eventloop     Stopping the EventLoop
[2023-08-22 11:29:38.903 (UTC-0500)] info/network       MQTT    | Shutting down the ConnectionManager
[2023-08-22 11:29:38.903 (UTC-0500)] info/network       ETH     | Shutting down the ConnectionManager
[2023-08-22 11:29:38.903 (UTC-0500)] info/network       UDP     | Shutting down the ConnectionManager
[2023-08-22 11:29:38.903 (UTC-0500)] info/network       TCP     | Shutting down the ConnectionManager
[2023-08-22 11:29:39.006 (UTC-0500)] info/eventloop     The EventLoop has stopped
```