open62541/open62541 #5844: UA_Client_forEachChildNodeCall with extended possibilities.

xydan83, 2023-06-28:
Hi!
ua_client_highlevel.c has UA_Client_forEachChildNodeCall. This is a good solution for browsing nodes, but sometimes you need to set some parameters like **BrowseDirection** or **includeSubtypes** or **referenceTypeId** for filtering. Can we create a function like UA_Client_forEachChildNodeCall_Ex for these parameters?

For example:
```C
UA_StatusCode UA_Client_forEachChildNodeCall_Ex(UA_Client* client, UA_NodeId parentNodeId, UA_BrowseDirection direction,  bool includeSubtypes, UA_NodeId ref_type, UA_NodeIteratorCallback callback, void* handle)
{
...
}
```

Can I create a PR for this?

jpfr, 2023-06-29:
Good idea.
We may not anticipate all options today.
Please think of a „option structure“ as the last argument.
With sane defaults of the option structure is NULL.

xydan83, 2023-06-29:
> Good idea. We may not anticipate all options today. Please think of a „option structure“ as the last argument. With sane defaults if the option structure is NULL.

Ok). We can change available function (without the new one with **_Ex**) but only with the default parameter, for example:

```
UA_StatusCode UA_Client_forEachChildNodeCall(
    UA_Client* client,
    UA_NodeId parentNodeId,
    UA_NodeIteratorCallback callback,
    void* handle,
    UA_BrowseDirection direction = UA_BROWSEDIRECTION_BOTH,
    bool includeSubtypes = false,
    UA_NodeId refType = UA_NODEID_NULL)
{
...
}
```
This way we will not break compatibility with the original version of the function (what if someone is already using it?), and at the same time we will be able to use more options.

The only thing that confuses me is if we make a deep copy of **parentNodeId**, why don't we pass it by a constant pointer?
[image]

xydan83, 2023-06-30:
Oh god, sorry, I forgot that we can't have default parameters in C because my main stack is C++))
I prepared some solution. Look here please.

xydan83, 2025-07-07:
Hi!) Please check)) Still waiting)