open62541/open62541 #5858: Unexpected session ID in async method processing on server

metint, 2023-07-04:
open62541 currently provides a way to process method calls asynchronously as explained in the [documentation](https://www.open62541.org/doc/1.3/server.html#id7). But using that approach leads to wrong session ID passed to the callback method attached to the OPC UA method.

In the [example code provided](https://github.com/open62541/open62541/blob/master/examples/tutorial_server_method_async.c), there is a call to [`UA_Server_call`](https://github.com/open62541/open62541/blob/master/examples/tutorial_server_method_async.c#L183) like below in the worker thread to get result for the OPC UA method call.

```
UA_CallMethodResult response = UA_Server_call(globalServer, &request->callMethodRequest);
```

If the implementation of `UA_Server_call` is traced in file [`ua_services_method.c`](https://github.com/open62541/open62541/blob/master/src/server/ua_services_method.c#L530) it is evident that admin session ID is passed instead of the session ID of the original client initiating the method call like below.

```
Operation_CallMethod(server, &server->adminSession, NULL, request, &result);
```

Is this intended or should be considered as a bug?

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.3.5
 - [X] Other OPC UA SDKs used (client or server): None
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2023-07-26:
+1

If this is intended, is there a way to a) get the session of the caller when dequeuing the next async request (with `UA_Server_getAsyncOperationNonBlocking`) and b) to do a `UA_Server_call` but also give the session as an argument?

jpfr, 2023-07-27:
@andreasebner , you and the team are currently looking at async processing.
Please consider this.