open62541/open62541 #5861: Unexpected result code for UA_Client_connect,when using method in example client_connect_loop.c

sampsonsxz, 2023-07-06:
## Description
In the [example code provided](https://github.com/open62541/open62541/tree/master/examples/client_connect_loop.c),there is a method to reconnect to server,the following is my code 

```c++
void opcclientwrapper::initClient()
{
    client = UA_Client_new();
    cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);
    cc->timeout = 1000;
}

void opcclientwrapper::runClient()
{
    while(running)
    {
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.9.192:14840");
        if(retval != UA_STATUSCODE_GOOD)
        {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Not connected. Retrying to connect in 5 second status : %x\n",retval);
            UA_sleep_ms(1000);
            continue;
        }
        if(bFirstConnect) {
            UA_NodeId targetNodeId = UA_NODEID_STRING(2, (char*)"Unit1_Ia");
            addMonitoredItemToVariable(client, &targetNodeId);
        }
        if(retval == UA_STATUSCODE_GOOD) {
            // send publish request, time out 1000ms
            UA_Client_run_iterate(client, 500);
            QCoreApplication::processEvents(QEventLoop::AllEvents, 100);
        }
    }
}
```
For the first loop I got the "retval = UA_STATUSCODE_GOOD" ,but the next loop I have got 0x800A0000 "retval =  UA_STATUSCODE_BADTIMEOUT " and the log shows that "Client already connected",Is this is a bug?

## Background Information / Reproduction Steps
I debug my code  , I found that " client->discoveryUrl.length is zero".When I call the function UA_Client_connect, the first step is to execute "UA_String_clear(&client->discoveryUrl);" I guess that's why the  client->discoveryUrl.length = 0.
```c
static UA_StatusCode
connectSync(UA_Client *client) {
    UA_DateTime now = UA_DateTime_nowMonotonic();
    UA_DateTime maxDate = now + ((UA_DateTime)client->config.timeout * UA_DATETIME_MSEC);

    UA_StatusCode retval = initConnect(client);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;

    while(retval == UA_STATUSCODE_GOOD) {
        if(!client->endpointsHandshake && !client->findServersHandshake &&
           client->discoveryUrl.length > 0 &&
           (client->sessionState == UA_SESSIONSTATE_ACTIVATED ||
            (client->noSession && client->channel.state == UA_SECURECHANNELSTATE_OPEN)))
            break;
        now = UA_DateTime_nowMonotonic();
        if(maxDate < now)
            return UA_STATUSCODE_BADTIMEOUT;
        retval = UA_Client_run_iterate(client,
                                       (UA_UInt32)((maxDate - now) / UA_DATETIME_MSEC));
    }

    return retval;
}
```

Here is the logging for mycode :
```
2023-07-06 01:14:58.021 (UTC-0700)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-07-06 01:14:58.038 (UTC-0700)] info/channel	Connection 4 | SecureChannel 11 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-07-06 01:14:58.038 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-07-06 01:14:58.048 (UTC-0700)] info/client	Use the EndpointURL opc.tcp://HTNICE:14840/ returned from FindServers
[2023-07-06 01:14:58.048 (UTC-0700)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2023-07-06 01:14:58.053 (UTC-0700)] info/channel	Connection 4 | SecureChannel 12 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-07-06 01:14:58.053 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-07-06 01:14:58.055 (UTC-0700)] info/client	Selected endpoint 0 in URL opc.tcp://HTNICE:14840/ with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-07-06 01:14:58.055 (UTC-0700)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-07-06 01:14:58.057 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-07-06 01:14:58.059 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-07-06 01:14:58.061 (UTC-0700)] info/client	Create subscription succeeded, id 7

[2023-07-06 01:14:58.105 (UTC-0700)] info/client	Monitoring 'Unit1_Ia', id 1

[2023-07-06 01:14:58.572 (UTC-0700)] info/client	Received Notification
[2023-07-06 01:14:58.572 (UTC-0700)] info/client	SubId:7, MonId:1, Current Value: 401.674643

[2023-07-06 01:14:58.572 (UTC-0700)] warn/client	Client already connected
[2023-07-06 01:14:59.572 (UTC-0700)] error/client	Not connected. Retrying to connect in 5 second status : 800a0000

[2023-07-06 01:15:00.572 (UTC-0700)] warn/client	Client already connected
[2023-07-06 01:15:01.573 (UTC-0700)] error/client	Not connected. Retrying to connect in 5 second status : 800a0000
```

Used CMake options:

```bash
cmake -DUA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):1.3.6
 - [x] Other OPC UA SDKs used (client or server): OPCUAServer:QUaServer
 - [x] Operating system:Ubuntu20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

aiweed, 2023-09-12:
The problem I encountered was similar to yours. When I commented 
```
UA_String_clear(&client->endpointUrl);
UA_String_clear(&client->discoveryUrl);
```
 out these two lines of code, it started working well. However, I’m not sure if there are any other issues.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.