open62541/open62541 #5867: Want to read an array in my server after the client has changed values in the array

SmallfaceR, 2023-07-11:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.32
 - [x ] Other OPC UA SDKs used (client or server): server
 - [x] Operating system: Linux
 - [] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [] Wireshark network dump attached
 - [] Self-contained code example attached
 - [] Critical issue

Hello I need help with the callback routine onwrite. I have a node with values in an array:
[image]

Then i have the callback routine:
[image]

I use UAExpert the write the array, for this i change the second value to 65 and push the write button. In my callback routine I alwys get the first value of the array when I use the "UA_Server_readValue", function. How can I get the hole array read out after the client changed the values?
[image]

SmallfaceR, 2023-07-13:
Anyone has an idea?

SmallfaceR, 2023-07-13:
Its okay I found a way: 
for the first element:                                  UA_Double *pointer = (UA_Double *) opc_value.data;
for the next i elements:                              pointer = (UA_Double *) opc_value.data+i;

geludwig, 2024-02-12:
> Its okay I found a way: for the first element: UA_Double *pointer = (UA_Double *) opc_value.data; for the next i elements: pointer = (UA_Double *) opc_value.data+i;

This did not work for me, because `(UA_Double *) opc_value.data+i` only increments the value by `i` and does not output the value on address `i`.
To access the values in the data array, I incremented the pointer address: `*++(UA_Byte*)val.data`.

```
#define NODE_MAX_SIZE 1024

int nodeRead(UA_Client* (*client)) {

    /* Init node items */
    UA_ReadValueId item[NODE_MAX_SIZE];

    /* Number of nodes to read */
    int nodeCount = 2;

    /* Node 0 */
    UA_ReadValueId_init(&item[0]);
    item[0].attributeId = UA_ATTRIBUTEID_VALUE;
    item[0].nodeId = UA_NODEID_NUMERIC(4, 10);

    /* Node 1 */
    UA_ReadValueId_init(&item[2]);
    item[1].attributeId = UA_ATTRIBUTEID_VALUE;
    item[1].nodeId = UA_NODEID_NUMERIC(4, 11);

    /* Request and response */
    UA_ReadRequest request;
    UA_ReadRequest_init(&request);
    request.nodesToRead = &item;
    request.nodesToReadSize = nodeCount;
    UA_ReadResponse response = UA_Client_Service_read(*client, request);

    /* Get value from node i */
    for (int i = 0; i < nodeCount; i++) {
        UA_DataValue* nodeResult = &response.results[i];
        UA_Variant* nodeValue = UA_Variant_new();
        memcpy(nodeValue, &nodeResult->value, sizeof(UA_Variant));
        UA_Variant_init(&nodeResult->value);

        /* Get byte array (or do for loop here) */
        printf("val: %i\n", *(UA_Byte*)nodeValue->data);
        printf("val: %i\n", *++(UA_Byte*)nodeValue->data);
        printf("val: %i\n", *++(UA_Byte*)nodeValue->data);
    }

    return 0;
}
```