open62541/open62541 #5878: There is no security policy Aes128_Sha256_RsaOaep

guoh27, 2023-07-14:
## Description

There is no security policy Aes128_Sha256_RsaOaep

## Background Information / Reproduction Steps

When I use http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep as securityPolicyUri,  I got `No suitable endpoint found` error

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.6
 - [ ] Other OPC UA SDKs used (client or server): open62541
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

GoetzGoerisch, 2023-07-14:
Currently being worked on here: https://github.com/open62541/open62541/pull/5512

guoh27, 2023-07-14:
Thanks a lot, I'll be following its progress.

jpfr, 2023-07-14:
Done.

closed 2023-07-14 by jpfr