open62541/open62541 #5881: Enabled MQTT, but protocol is not supported (tutorial_pubsub_mqtt_publish.c)

starcore2022, 2023-07-17:
## Description
When trying to run the tutorial included for MQTT on Linux (Ubuntu 22.04), the system throws error. The error says that the requested protocol is not supported I enabled the MQTT option while cmake.

I would be thankful, if someone could help me out with it.

## Background Information / Reproduction Steps
Because of a project needing to be put on a final version of OPC UA, I reinstalled open6254. I changed the git branch to 1.4 (tried with 1.2, 1.3, 1.3.3.1, with the same error). The installation is successful, but whether the MQTT option is enabled or not it is finds that the protocol is not supported. Even though it worked with the February master branch.

I built everything from source:

git clone <open62541>
cd open62541
git switch 1.4
mkdir build && cd build
ccmake ..
make
sudo make install
cd bin/examples
./tutorial_pubsub_mqtt_publish

Used CCMake options:

```
ENABLE_EXAMPLES ON
UA_BUILD_EXAMPLES ON
UA_ENABLE_HARDENING ON
UA_ENABLE_JSON_ENCODING ON
UA_ENABLE_METHIDCALLS ON
UA_ENABLE_MQTT ON
UA_ENABLE_NODEMANAGEMENT ON
UA_ENABLE_NODESETLOADER ON
UA_ENABLE_PUBSUB ON
UA_ENABLE_PUBSUB_DELTAFRAMES ON
UA_ENABLE_PUBSUB_MONITORING ON
UA_FORCE_WERROR OFF
```

ERROR LOG:
```
[2023-07-17 12:26:06.049 (UTC+0200)] info/eventloop	Starting the EventLoop
[2023-07-17 12:26:06.049 (UTC+0200)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-07-17 12:26:06.049 (UTC+0200)] info/server	AccessControl: Anonymous login is enabled
[2023-07-17 12:26:06.049 (UTC+0200)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2023-07-17 12:26:06.049 (UTC+0200)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2023-07-17 12:26:06.053 (UTC+0200)] warn/server	Writing the value of Node i=12779 failed with the following reason: DataType of the value is incompatible
[2023-07-17 12:26:06.053 (UTC+0200)] info/session	TCP 0	| SC 0	| Session "Administrator"	| WriteRequest returned status code BadTypeMismatch
[2023-07-17 12:26:06.053 (UTC+0200)] info/session	TCP 0	| SC 0	| Session "Administrator"	| AddNode (i=12779): Could not create a default value with StatusCode BadTypeMismatch
[2023-07-17 12:26:06.053 (UTC+0200)] warn/server	Writing the value of Node i=2165 failed with the following reason: DataType of the value is incompatible
[2023-07-17 12:26:06.053 (UTC+0200)] info/session	TCP 0	| SC 0	| Session "Administrator"	| WriteRequest returned status code BadTypeMismatch
[2023-07-17 12:26:06.053 (UTC+0200)] info/session	TCP 0	| SC 0	| Session "Administrator"	| AddNode (i=2165): Could not create a default value with StatusCode BadTypeMismatch
[2023-07-17 12:26:06.055 (UTC+0200)] error/pubsub	Connection ns=1;i=1	| The requested protocol is not supported
[2023-07-17 12:26:06.055 (UTC+0200)] error/server	Error Name = BadNotFound
[2023-07-17 12:26:06.055 (UTC+0200)] info/server	PubSub cleanup was called.
[2023-07-17 12:26:06.055 (UTC+0200)] info/eventloop	Stopping the EventLoop
[2023-07-17 12:26:06.055 (UTC+0200)] info/network	MQTT	| Shutting down the ConnectionManager
[2023-07-17 12:26:06.055 (UTC+0200)] info/network	ETH	| Shutting down the ConnectionManager
[2023-07-17 12:26:06.055 (UTC+0200)] info/network	UDP	| Shutting down the ConnectionManager
[2023-07-17 12:26:06.055 (UTC+0200)] info/network	TCP	| Shutting down the ConnectionManager
[2023-07-17 12:26:06.154 (UTC+0200)] info/eventloop	The EventLoop has stopped

```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): -
 - [ ] Operating system: Linux - Ubuntu 22.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached

eayt, 2023-07-17:
I am having the same problem. Even when I built the library in version 1.3, I couldn't run the sample code. I got the error "No such file or directory ua_pubsub.h". In version 1.4, I encountered the same error as you. I've been trying to build the 1.3 version for about 1 week with the mqtt pubsub feature open. How did you build version 1.3? Can you share the cmake command?

starcore2022, 2023-07-17:
Hi! In your case I would suspect, that the PubSub is not enabled, or it being disabled when, changing git branch. I used ccmake, which is a GUI for cmake. I will look up the cmake command for it.

starcore2022, 2023-07-17:
Not everything is included in it, but should work. Successfully built it, after switching to 1.3, although the MQTT example is still doesn't run as expected for me.

The cmake command: 
cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DUA_ENABLE_PUBSUB=ON -DENABLE_EXAMPLES=ON -DUA_BUILD_EXAMPLES=ON -DUA_JSON_ENCODING=ON -DUA_ENABLE_MQTT=ON -DUA_FORCE_WERROR=OFF

eayt, 2023-07-17:
Thanks for your response it really helped me. I built successfully.  When I look at the terminal output, there seems to be no problem. PubSub MQTT: Publish message is going through terminal periodically. When I open a separate terminal with mosquitto_sub -t customTopic, the data flow can be observed.

closed 2023-07-17 by starcore2022

eayt, 2023-07-17:
And also for 1.3  -DUA_JSON_ENCODING=ON  must be changed to  -DUA_ENABLE_JSON_ENCODING=ON to gets meaningful output from mqtt subscriber.