open62541/open62541 #5882: Server constantly updates timestamp of monitored items with data change trigger = UA_DATACHANGETRIGGER_STATUSVALUETIMESTAMP

paddor, 2023-07-18:
## Description

## Background Information / Reproduction Steps

When I connect UaExpert (v1.6.3) to a Open62541 Server (v1.3.6) and drag a variable (e.g. `i=2266 [Server_ServerStatus_BuildInfo_BuildDate]`) into the Data Access View and change the Monitored Item Settings so that Data Change Trigger is `Status/Value/Timestamp`, both Source Timestamp and Server Timestamp keep getting updated.

This also happens to my own variables (added dynamically, not imported from a nodeset). I tried with a writable Bool variable node. Interestingly, as soon as I write into the variable using UaExpert, the updating of the timestamps stops.

Using Ubuntu Linux 22.04.

Used CMake options:

```
-DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g'
```

paddor, 2023-07-18:
To reproduce, I compiled the library:

```
git clone https://github.com/open62541/open62541.git
cd open62541
git submodule update --init --recursive
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Debug  -DUA_ENABLE_DIAGNOSTICS=ON -DUA_NAMESPACE_ZERO=FULL ..
make
sudo make install
sudo ldconfig
```

And then just run the example program https://github.com/open62541/open62541/blob/master/examples/tutorial_server_variable.c.
```
gcc -std=c99 tutorial_server_variable.c -lopen62541 -o tutorial_server_variable
./tutorial_server_variable
```

Connect with UaExpert, expand `ServerDiagnostics`, drag and drop the `BuildInfo`, set to `Status/Value/Timestamp`, and watch the timestamp continually get updated.

paddor, 2023-07-18:
More findings:

* `detectVAlueChange()` seems to behave correctly, but the `value.sourceTimestamp` just keeps getting updated
* even read operations update the timestamps (Refresh button in UaExpert)
* it's not just a bug in UaExpert. WireShark shows the variables repeatedly in `PublishResponse` messages.
* the variable `the.answer` from the `tutorial_server_variable.c` example works correctly because of `UA_Server_writeValue()`  call
* `UA_Server_writeValue()` seems to have a side-effect that stops the continuous updating of timestamps
* this phenomenon also shows on master branch

jpfr, 2023-07-19:
I think the current behavior is as follows:

the server timestamp, if requested, is always the current time.

The source timestamp is used as defined by the source. If no source timestamp is defined we “fill in” the current time.

So when you filter against the timestamp, and we fill in the current time, you get very fast updates.

sounds correct. How do you propose the behavior should be different?

paddor, 2023-07-19:
Thanks for your response. To be honest, this behavior is quite confusing and doesn't make sense in combination with `UA_DATACHANGETRIGGER_STATUSVALUETIMESTAMP`. If one wants to subscribe to changes in either Status, Value, or Timestamp, one wouldn't expect to be flooded with updates with no writes to the node. It's okay if the `SourceTimestamp` is filled in by the server, but constantly updating it does not make sense in our opinion (and our project's requirements).

This is the behavior of the Unified Automation SDK: When a node (like `Server_ServerStatus_State`) is refreshed, the `ServerTimestamp` is updated, but the `SourceTimestamp` stays the same. See screenshot:
![image](https://github.com/open62541/open62541/assets/8876/a46a3d24-1b51-47d7-b3ef-83fb
346ff868)

arohr, 2023-07-28:
> I think the current behavior is as follows:
>
> the server timestamp, if requested, is always the current time.
>
> The source timestamp is used as defined by the source. If no source timestamp is defined [...]

In our observation the problem is that the `sourceTimestamp` set by the application code seems not to be respected and gets then replaced with the current time, which leads to fast updates if the client has set the data change trigger to `Status/Value/Timestamp` although neither the value nor the timestamp of the variable in the process has changed. This appears to be a bug to me.

paddor, 2023-08-14:
@jpfr Neither setting `sourceTimestamp` nor setting `serverTimestamp` on a UA imported from a nodeset stop the fast updates.

Please advise on how to work around this behavior.

jpfr, 2023-08-15:
It is possible that we use the "current time" for the timestamp only if that information was not explictly set before.
I think what happens is that when you write the variable an explicit timestamp gets written.
Try to initialize your variables with a timestamp.

If that solves your case, then we only need better documentation.
Otherwise we need to make a thoughtful change to the logic - and avoid breaking applications relying on previous behavior.

paddor, 2023-08-15:
As far as I know it's impossible to initialize variables with a certain value during the import of an XML nodeset. Explicitly setting just the timestamps alone (using `UA_Server_writeDataValue()`) after the import does not help stop the update flooding.

Our current workaround is to not use the nodeset import functionality, and instead use `UA_Server_addVariableNode()` followed by a `UA_Server_writeDataValue()` to set the `sourceTimestamp`, which avoids the update flooding.

paddor, 2024-02-19:
Just curious, any updates here?