open62541/open62541 #5888: Debian packaging bookworm pdflatex error - package dependency is missing

debdanfli, 2023-07-19:
## Description

Trying to build a debian package for bookworm in a clean pbuilder environment. Ends with the following pdflatex error: 

```bash
[100%] Generating PDF documentation from LaTeX sources
cd /home/fli/testenvpbuilder/project/open62541/obj-x86_64-linux-gnu/doc_latex && /usr/bin/pdflatex -interaction=batchmode open62541.tex
This is pdfTeX, Version 3.141592653-2.6-1.40.24 (TeX Live 2022/Debian) (preloaded format=pdflatex)
 restricted \write18 enabled.
entering extended mode
make[5]: *** [doc/CMakeFiles/doc_pdf.dir/build.make:74: doc/CMakeFiles/doc_pdf] Error 1
make[5]: Leaving directory '/home/fli/testenvpbuilder/project/open62541/obj-x86_64-linux-gnu'
make[4]: *** [CMakeFiles/Makefile2:605: doc/CMakeFiles/doc_pdf.dir/all] Error 2
make[4]: Leaving directory '/home/fli/testenvpbuilder/project/open62541/obj-x86_64-linux-gnu'
make[3]: *** [CMakeFiles/Makefile2:612: doc/CMakeFiles/doc_pdf.dir/rule] Error 2
make[3]: Leaving directory '/home/fli/testenvpbuilder/project/open62541/obj-x86_64-linux-gnu'
make[2]: *** [Makefile:361: doc_pdf] Error 2
make[2]: Leaving directory '/home/fli/testenvpbuilder/project/open62541/obj-x86_64-linux-gnu'
make[1]: *** [debian/rules:15: override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/fli/testenvpbuilder/project/open62541'
make: *** [debian/rules:22: build] Error 2
```

Finding the problem with changing the pdflatex interaction mode to: errorstopmode 

The pbuilder output with the new mode said this: 

```bash
(/usr/share/texlive/texmf-dist/tex/generic/babel/babel.sty
(/usr/share/texlive/texmf-dist/tex/generic/babel/txtbabel.def)
(/usr/share/texlive/texmf-dist/tex/generic/babel-english/english.ldf))
(/usr/share/texlive/texmf-dist/tex/generic/babel/locale/en/babel-english.tex)

! LaTeX Error: File `tgtermes.sty' not found.

Type X to quit or <RETURN> to proceed,
or enter new name. (Default extension: sty)

Enter file name: 
! Emergency stop.
<read *> 
         
l.38 \usepackage
                {tgheros}^^M
!  ==> Fatal error occurred, no output PDF file produced!
Transcript written on open62541.log.
```

The `tgtermes.sty` file comes from the debian package `tex-gyre`.

Fix the Bug for me:
Adding the missing debian package "tex-gyre" to the debian/control "Build-Depends" list. 

It seems to me that some build-depends of some debian package in "Build-Depends" file has changed in bookworm. 

## Background Information / Reproduction Steps

Using the pdebuild Tool for package building - https://manpages.debian.org/jessie/pbuilder/pdebuild.1.en.html

The branch pack/v1.3.6 works with debian bullseye very well, but not with bookworm. 

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): pack/v1.3.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: amd64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-07-19:
Good find!
Would you prepare a PR?

debdanfli, 2023-07-20:
Ok. 
Hope that is the right path, tools/packaging/debian/control-template.
Fork the branch for preparation, can you use this fork?

sergiomb2, 2024-07-13:
for centos stream + epel 9 we need `dnf install "tex(pdflscape.sty)" `

o6-nbeier, 2026-10-22:
Closing for no activity.

closed 2026-10-22 by o6-nbeier