open62541/open62541 #5915: EventLoop(UPD): Server binds socket to remote host address

keba-estr, 2023-08-01:
## Description

The Address of the PubSub configuration contains an Url and a NetworkInterface. As I see it, the Url is the remote part including some remote URL and a port, whereas the NetworkInterface is the local network interface the connection should send to and receive from.

Opening a PubSub connection fails with the error trace "Error binding the socket to the address (Cannot assign requested address), closing".

Example:
Url: opc.udp://192.168.10.1:4840
NetworkInterface: 192.168.10.2
where 192.168.10.1 is the remote server url and 192.168.10.2 is the ip address of the local network interface.

## Background Information / Reproduction Steps

The function UA_PubSubConnection_connectUDP passes these parameters as key-value-pairs to the EventLoop. In the UDP EventLoop the "address" is read in UDP_openReceiveConnection and the resolved hostname is passed to UDP_registerListenSockets, where the function bind is called for this hostname.
According to the documentation of bind this function should be called with a local address, not with a remote address.

Maybe the parameter "interface" should be used instead here.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.4 branch + master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-08-02:
Are you on linux? Try to use the network interface name (such as eth0).

If that doesn't confirm to the OPC UA standard we will change that behavior.

keba-estr, 2023-08-02:
> Are you on linux? Try to use the network interface name (such as eth0).
> 
> If that doesn't confirm to the OPC UA standard we will change that behavior.

If I change the NetworkInterface from 192.168.10.2 to eth1, the behaviour does not change, because the open62541 evaluates the Url for calling bind.

keba-estr, 2023-08-02:
If i use the parameter "interface", at least the configuration is successful.

keba-estr, 2023-08-02:
See PR #5919

keba-estr, 2023-08-22:
@jpfr: The following part of the specification is not met with the open62541 1.4 branch:

> The hostname for the Url in the PubSubConnection Address parameter is set to ‘localhost’ since the source address
is not used for filtering. The NetworkInterface field in the Address is not required and is only
configured if the Subscriber should listen only on the configured interface. If the
NetworkInterface is null or empty, the Subscriber should listen on all interfaces.

If the hostname ist set to "localhost" or "127.0.0.1", the subscriber only listens to the localhost interface and not to all interfaces.

jpfr, 2023-08-22:
> If the hostname ist set to "localhost" or "127.0.0.1", the subscriber only listens to the localhost interface and not to all interfaces.

We could change the behavior so that localhost, 127.0.0.1 and "any IPv6 address for localhost" are used to trigger "listen on all interfaces".

I think right now only "localhost" does that.

Works for you?

keba-estr, 2023-08-22:
> "localhost"

At the moment only an empty hostname triggers "listen on all interfaces" for the UDP implementation. The change would be fine for me. The host name is evaluated correctly in this case.
Is there a combination possible that the hostname is "localhsot", but the networkinterface specifies some specific network for receiving? Or is this against the specification?

closed 2024-02-19 by keba-estr