open62541/open62541 #5931: Storing dynamically generated nodes (objects/variables/variabletypes/datatypes/...)

merlin81, 2023-08-08:
A big thanks to the developers in advance, great framework !

Is there a recommended way to backup/restore nodes which are generated while runtime. I haven't found a hint in the documentation or on git.

I'm working on a "highly available"-server, so the server shouldn't be rebooted to get a node in, like modeler->xml nodeset->server reboot to read in. I've already implemented adding new nodes to the server by client as shown in the documentation, also the historizing looks good. 

Now i'm looking for a way to get these custom nodes persistent over crash/maintenance restarts.

thinking about ...
- afterAddNode-callback on server to backup the node
- export-functionality (node -> get struct/description -> save to xml/db)

undroidwish, 2023-08-10:
Maybe one of my latest experiments in loading/storing an open62541 address space using an SQLite database could serve you as a source of inspiration. The code is [here](https://www.androwish.org/home/file?name=jni/topcua/library/sqlmodel.tcl&ci=tip). The ERD of the database schema is [here](https://www.androwish.org/home/file?name=jni/topcua/doc/sqlmodel.png&ci=tip). I shall admit that it is written mostly in unpopular, undead,  underrated Tcl.

merlin81, 2023-08-11:
thx @undroidwish ... so there is no common way to export the nodetree via plugin or library ? do you have a hint for a callback after the node is added while runtime ?

undroidwish, 2023-08-11:
Yes, watch out for `UA_ServerConfig.nodeLifeCycle` and implement something reasonable for the `constructor` and optionally `destructor` function pointers.

merlin81, 2023-08-15:
one question left to get this job done, looks good ... how can i get the parentnode of the new node in the constructor of UA_ServerConfig.nodeLifeCycle ?

**nodecontext seems to be reserverd for "user context" and the parent-children relationsship is stored in the parentnode (ref:hascomponent), not in the childnode right ?

undroidwish, 2023-08-15:
Not a cheap task indeed. You have the NodeId which is assumed to be the only knowledge of the thing which was constructed. You need to chase for inverse (hierarchical?) references of the node. If there's one, you've got the parent 🙌. If there are two, you found mom and dad 👨‍👩‍👦. If there are even more, perform paternity tests 😉.

merlin81, 2023-08-16:
to prevent children have to grow up without or with too many parents, let me clarify the question:

is there a way to grab the `parentNodeId` from `UA_[Client/Server]_addNode` (see below) while the `constructor` of  `UA_ServerConfig.nodeLifeCycle` is executed ?

- ` __UA_Client_addNode(UA_Client *client, const UA_NodeClass nodeClass, const UA_NodeId requestedNewNodeId, const UA_NodeId parentNodeId, ...)`

- `__UA_Server_addNode(UA_Server *server, const UA_NodeClass nodeClass, const UA_NodeId *requestedNewNodeId, const UA_NodeId *parentNodeId, ...)`

undroidwish, 2023-08-16:
I understand your question as an RFE (Request For Enhancement) to the `UA_GlobalNodeLifecycle` methods.

What are the feelings of the chief developers?