open62541/open62541 #5947: where is definition of  function "UA_DateTime_toJSON"

Jimmycaojp, 2023-08-16:
https://github.com/open62541/open62541/blob/66e4f6aa5bb0ac28d739d6ec258fca3a0967f3c8/src/ua_types_encoding_json.c#L61

I used UA_ENABLE_AMALGAMATION to configure cmake
and it generate a open62541.c&open62541.h. but it can't compile.
8>open62541.c
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240407,11): error C2054: 在“﻿”之后应输入“(”
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240407,11): error C2085: “UA_DateTime_toJSON”: 不在形参表中
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240408,1): error C2085: “ByteString_encodeJson”: 不在形参表中
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240411,1): error C2085: “writeChar”: 不在形参表中
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240411,33): error C2143: 语法错误: 缺少“;”(在“{”的前面)
8>D:\workprj\svn\IEC61499\uap_v1\soft\trunk\source\opcua\open62541\build\open62541.c(240435,12): warning C4013: “writeChar”未定义；假设外部返回 int

actually I can't the define of "UA_DateTime_toJSON", there is  only just a forward declaration
Is there anyone know this problem? or how to resolve it? 
thanks

jpfr, 2023-08-21:
You can use this method:
https://github.com/open62541/open62541/blob/bf3b5945da2d744706d626fdf8ca42b39e5ee7ed/include/open62541/types.h#L1325

The internal definition is an oversight.
We'll remove it.

Jimmycaojp, 2023-08-23:
thanks for help.
that open6241.c is generated automatically by cmake.
but as you said, I will remove this funtion declaration manually.
thank you again.