open62541/open62541 #5955: Legend for FEATURES.md

legas7, 2023-08-17:
Hi,

Could someone please help me understand what different 'moons' mean in [FEATURES.md](https://github.com/open62541/open62541/blob/master/FEATURES.md)

Best,
K.

starcore2022, 2023-08-18:
Hi! I am not completely sure, because I can't find the legend for it, but the new moon sign should be for newly added in features that may not be fully functional yet, mos likely a minimal functional version. While the full moon should be a fully completed feature that may or may not change before it is approved of it being final.
Have a nice day!

jpfr, 2023-09-25:
An update is provided in #6021.
I don't think we need a full legend, should be self-explanatory...