open62541/open62541 #5956: Tools directory of installed library is wrong

ccvca, 2023-08-18:
## Description
In https://github.com/open62541/open62541/commit/7547a8a9dc0b64eaf7a568305d82a78da6a6a6a1#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aL25
`set(open62541_TOOLS_DIR "${PROJECT_SOURCE_DIR}/tools")` moved from `CMakeLists.txt` to `/tools/cmake/macros_public.cmake`. 

https://github.com/open62541/open62541/blob/bd994c1609478e6cf2732b3671c55ab459cc8974/tools/cmake/macros_public.cmake#L1
When importing the library with `find_package(open62541 REQUIRED)` this results in a tools directory like this: `/home/user/UmatiServerDev/open62541_update/Sample-Server/tools`.

Then an error like this appears:
```
gmake[5]: *** Keine Regel vorhanden, um das Ziel „../../tools/nodeset_compiler/nodeset_compiler.py“, 
  benötigt von „../../src_generated/namespace_irdi_generated.c“, zu erstellen.  Schluss.
```

The path is correctly set in the Config-Script here:
https://github.com/open62541/open62541/blob/bd994c1609478e6cf2732b3671c55ab459cc8974/tools/cmake/open62541Config.cmake.in#L5

```
set (open62541_TOOLS_DIR ${PACKAGE_PREFIX_DIR}/share/open62541/tools CACHE PATH "Path to the directory that contains the tooling of the stack")
```

but as the macros are included, the wrong path is used later on.

## Possible fixes

- Move `set (open62541_TOOLS_DIR ...` to CMakeLists.txt again
- Check if open62541_TOOLS_DIR is already set before setting the 'fallback' `${PROJECT_SOURCE_DIR}/tools`

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): bd994c1609478e6cf2732b3671c55ab459cc8974
 - [x] Operating system: Debian
 - [x] Critical issue

GoetzGoerisch, 2023-09-11:
@jpfr could you please have a look?

jpfr, 2023-09-25:
I prefer option No. 2

PR #6049 (2023-10-09): fix(core): Do not reset tools directory when installed
Fix: #5956
closed 2023-10-09 by jpfr