open62541/open62541 #5977: PubSub MQTT examples exit with "The requested protocol is not supported"

MarkoMuc, 2023-08-24:
## Description
On version 1.4, PubSub MQTT examples fail with "The requested protocol is not supported", even when MQTT is enabled at build.

## Background Information / Reproduction Steps

- Build project with MQTT and PubSub enabled. 
- Run the MQTT examples (either tutorial_pubsub_mqtt_subscribe or tutorial_pubsub_mqtt_publish) 
- The examples exit with error "The requested protocol is not supported"

Used CMake options:

```bash
cmake -DUA_LOGLEVEL=100 -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_MQTT=ON -DUA_BUILD_EXAMPELS=ON -DUA_ENABLE_JSON_ENCODING=ON ..
```

## Checklist

Please provide the following information:

 - [ - ] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ - ] Operating system: Fedora 38
 - [ - ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue
[tutorial_pubsub_mqtt_subscribe_log.txt](https://github.com/open62541/open62541/files/12426344/tutorial_pubsub_mqtt_subscribe_log.txt)

tukotech, 2023-08-25:
@MarkoMuc  - could you please try to change the following line - 
https://github.com/open62541/open62541/blob/3ad448ec722087a369c8b2c94391dfe944679ed4/examples/pubsub/tutorial_pubsub_mqtt_publish.c#L50

Into this one:
```
#define TRANSPORT_PROFILE_URI        "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-uadp"
```

MarkoMuc, 2023-08-25:
@tukotech I changed it and now I don't get the error anymore. Thank you for the help.

closed 2023-08-25 by MarkoMuc