open62541/open62541 #5984: Server stops with Segmentation Fault when we configure wrong security policy in PubSub subscriber

jchirantan, 2023-08-25:
## Description
If we configure subscriber with wrong policy than publisher(AES128CTR/AES256CTR), server gets stopped with segmentation fault.

Can you add some error handling mechanism and just stop the connection without stopping server?

## Background Information / Reproduction Steps
Step 1: Start pubsub_publish_encrypted.exe with security mode as Sign and Encrypt and security Policy as AES128CTR
Step 2: Start pubsub_subscribe_encrypted.exe with security mode as Sign and Encrypt and security Policy as AES256CTR
OR Vice-versa
After few seconds of warnings, we get segmentation fault and server stops

[image]

### Linux:
[image]

### Windows:
[image]

[image]

Used CMake options:

cmake.exe .. -G "Visual Studio 17 2022" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=OFF -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_DEBUG_SANITIZER=OFF -DUA_ENABLE_DISCOVERY_SEMAPHORE=OFF -DUA_ENABLE_HARDENING=OFF -DUA_ENABLE_NODEMANAGEMENT=OFF -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS=OFF -DUA_ENABLE_PARSING=OFF -DUA_ENABLE_PUBSUB_DELTAFRAMES=OFF -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=OFF -DUA_ENABLE_STATUSCODE_DESCRIPTIONS=OFF -DUA_ENABLE_SUBSCRIPTIONS=OFF -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=OFF -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_FORCE_WERROR=OFF -DUA_MSVC_FORCE_STATIC_CRT=OFF -DUA_ENABLE_ENCRYPTION_TPM2=OFF -DUA_LOGLEVEL=400 -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_PUBSUB_ENCRYPTION=ON 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.6
 - [x] Operating system: Windows and Linux
 - [x] UA_LOGLEVEL=400
 - [x] Critical issue

max65482, 2023-12-05:
I could reproduce this with tag 1.3.6, but not with the latest master. Seems to be fixed.

jchirantan, 2023-12-13:
Thanks @max65482 for the information, 
- I tried to reproduce this issue with latest tag 1.3.9.
- Now the server is stopping a bit late i.e., after 10-15 minutes of continuous running
- Earlier it was stopping after 1-2 minutes.
- Can we fix this problem such that, the server will never stop and run unless stopped externally?

max65482, 2023-12-13:
I could now reproduce your issue with 1.3.9 after just 43 seconds.

This is my stack trace:
```
pubsub_subscribe_encrypted.exe!Variant_clear(UA_Variant * p, const UA_DataType * _) Line 544	C
pubsub_subscribe_encrypted.exe!DataValue_clear(UA_DataValue * p, const UA_DataType * _) Line 971	C
pubsub_subscribe_encrypted.exe!UA_clear(void * p, const UA_DataType * type) Line 1301	C
pubsub_subscribe_encrypted.exe!UA_DataValue_clear(UA_DataValue * p) Line 725	C
pubsub_subscribe_encrypted.exe!UA_DataSetMessage_clear(const UA_DataSetMessage * p) Line 1729	C
pubsub_subscribe_encrypted.exe!UA_NetworkMessage_clear(UA_NetworkMessage * p) Line 1125	C
pubsub_subscribe_encrypted.exe!decodeAndProcessNetworkMessage(UA_Server * server, UA_ReaderGroup * readerGroup, UA_PubSubConnection * connection, UA_String * buffer) Line 1348	C
pubsub_subscribe_encrypted.exe!decodeAndProcessFun(UA_PubSubChannel * channel, void * cbContext, const UA_String * buffer) Line 1411	C
pubsub_subscribe_encrypted.exe!UA_PubSubChannelUDPMC_receive(UA_PubSubChannel * channel, UA_ExtensionObject * transportSettings, unsigned int(*)(UA_PubSubChannel *, void *, const UA_String *) receiveCallback, void * receiveCallbackContext, unsigned int timeout) Line 610	C
pubsub_subscribe_encrypted.exe!receiveBufferedNetworkMessage(UA_Server * server, UA_ReaderGroup * readerGroup, UA_PubSubConnection * connection) Line 1434	C
pubsub_subscribe_encrypted.exe!UA_ReaderGroup_subscribeCallback(UA_Server * server, UA_ReaderGroup * readerGroup) Line 639	C
pubsub_subscribe_encrypted.exe!serverExecuteRepeatedCallback(UA_Server * server, void(*)(void *, void *) cb, void * callbackApplication, void * data) Line 687	C
pubsub_subscribe_encrypted.exe!UA_Timer_process(UA_Timer * t, __int64 nowMonotonic, void(*)(void *, void(*)(void *, void *), void *, void *) executionCallback, void * executionApplication) Line 264	C
pubsub_subscribe_encrypted.exe!UA_Server_run_iterate(UA_Server * server, bool waitInternal) Line 693	C
pubsub_subscribe_encrypted.exe!UA_Server_run(UA_Server * server, const volatile bool * running) Line 789	C
pubsub_subscribe_encrypted.exe!run(UA_String * transportProfile, UA_NetworkAddressUrlDataType * networkAddressUrl) Line 335	C
pubsub_subscribe_encrypted.exe!main(int argc, char * * argv) Line 375	C
```

The crash is caused by a misplaced call of `UA_NetworkMessage_clear`. However, it seems to be fixed in the latest master:
https://github.com/open62541/open62541/blob/cc5c7da6a11bf0fe012dbcd5af48a4f1d9fc6297/src/pubsub/ua_pubsub_eventloop.c#L869-L887

jchirantan, 2023-12-17:
Thanks @max65482 for such a quick response.
I just built the master branch code and I can see the issue is fixed.

May I know the tentative release date of next tag of 1.4 which will contain this fix?

Thanks,
Chirantan

jchirantan, 2025-03-19:
Hi @max65482 @jpfr,

This issue is still present in latest release v1.4.10. 

Any updates on this?

o6-nbeier, 2026-10-20:
If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier