open62541/open62541 #5986: Build for QNX

PJ-Kowalski, 2023-08-26:
Hello
I'm trying to build library for QNX system, in Visual Studio Code, i modify CMakeList to
 include(Toolchain-QNX-7.1.0) 
and  tollchain file to have paths according to my local environment
and i execute from VS:   
```bash
cmake -G "Visual Studio 17 2022" -DUA_ARCHITECTURE=posix -DCMAKE_TOOLCHAIN_FILE=../tools/cmake/Toolchain-QNX-7.1.0.cmake ../
```
As  output i get files in build catalog with contain visualstudio solution file. 
Then I was trying to build it from vs but i get error: 
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1181	cannot open input file 'D:\Repos\open62541\build\open62541-object.dir\Release\ua_types.obj'	open62541 (open62541\lib\open62541)	D:\Repos\open62541\build\LINK	1	
Error	D8021	invalid numeric argument '/Wshadow'	open62541-object	D:\Repos\open62541\build\cl	1	
Error	D8021	invalid numeric argument '/Wshadow'	open62541-plugins	D:\Repos\open62541\build\cl	1	

Cmake cashe paths looks ok 
[image]

My process of compiling this library is ok, or i should do something wrong ? 
I need to build this libary for x86 and arm architecture

Best regards
Piotr

tukotech, 2023-08-28:
Not that it helps, just additional information.  QNX 7.1 does not support 32-bit for x86. 

https://www.qnx.com/developers/docs/7.1/#com.qnx.doc.neutrino.prog/topic/64bits.html

jpfr, 2023-09-14:
Are we adding the wshadow flag?
Please check the CMakeLists.txt.
I‘m sure you can edit the flags in VS after creating the project.

PJ-Kowalski, 2023-09-14:
I was trying with wshadow flag, but currently i was able to compile for QNX inside Momentics, for this purpose i use single relies files, its working fne under QNX

closed 2023-09-14 by PJ-Kowalski