open62541/open62541 #5989: UA_Client_Service_read-> UA_ReadResponse -> hasServerTimeStamp return false even if there is a server time stamp.

ffreality, 2023-08-30:
## Description
`UA_ReadResponse` which I got from `UA_Client_Service_read` doesn't return true with `results->hasServerTimestamp`

## Background Information / Reproduction Steps
I created an `Unreal Engine 5` client plugin and I want to read target node's properties including their timestamps.

I am using latest version of `Prosys Simulation Server Free Edition`. 
`UOPC_UA_Client` is an UObject for Unreal Engine
`FOPC_UA_Data_Infos` is a struct for Unreal Engine
Currently I can read values, name, data type, data name, node id string, description and  source time stamp

Prosys has server timestamp for its nodes and I can read it with `UA_HistoryData` which I can generate from `UA_Client_HistoryRead_raw`

But `UA_ReadResponse` from `UA_Client_Service_read` doesn't return true for hasServerTimestamp.
So, I can't read correct time.

you can see simplified codes about read and timestamps.

```
void OPC_Read::Read_OPC_UA_Value(UOPC_UA_Client* InConnection, UA_NodeId OPC_UA_NodeID, UA_AttributeId AttributeId, UA_ReadResponse& ReadResponse)
{
	UA_ReadRequest ValueReadRequest;
	UA_ReadRequest_init(&ValueReadRequest);

	UA_ReadValueId ValueReadRequestID[1];
	UA_ReadValueId_init(&ValueReadRequestID[0]);
	ValueReadRequestID[0].attributeId = AttributeId;
	ValueReadRequestID[0].nodeId = OPC_UA_NodeID;

	ValueReadRequest.nodesToRead = ValueReadRequestID;
	ValueReadRequest.nodesToReadSize = 1;

	ReadResponse = UA_Client_Service_read(InConnection->Client, ValueReadRequest);
}

UA_ReadResponse ReadResponse;
UA_ReadResponse_init(&ReadResponse);
OPC_Read::Read_OPC_UA_Value(InConnection, OPC_UA_NodeID, UA_AttributeId::UA_ATTRIBUTEID_VALUE, ReadResponse);

if (ReadResponse.results->status == UA_STATUSCODE_GOOD)
{
	FOPC_UA_Data_Infos DataInfos;

	if (ReadResponse.results->hasSourceTimestamp == true)
	{
		DataInfos.SourceTime = FDateTime::FromUnixTimestamp(UA_DateTime_toUnixTime(ReadResponse.results >sourceTimestamp));
	}

	if (ReadResponse.results->hasServerTimestamp == true)
	{
		DataInfos.ServerTime = FDateTime::FromUnixTimestamp(UA_DateTime_toUnixTime(ReadResponse.results->serverTimestamp));
	}
```

Is there any suggestion for this ?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Master Branch from August 23
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

undroidwish, 2023-08-30:
You might want to set the field

    UA_TimestampsToReturn timestampsToReturn;

of the read request to the things you wish for.

ffreality, 2023-08-31:
Thanks, it worked !

closed 2023-08-31 by ffreality