open62541/open62541 #5991: UA_LOGLEVEL is set to 100 but not seeing Trace and Debug in the log

tukotech, 2023-09-01:
## UA_LOGLEVEL is set to 100 but not seeing Trace and Debug - I can see INFO/WARN/ERROR. 

## Background Information / Reproduction Steps
I can step through the debugger and see that TRACE and ERROR logs are hit but I am not seeing the text in the console. 

Used CMake options:
```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_MULTITHREADING=100 -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_PUBSUB=ON -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_MQTT=ON -DUA_ENABLE_PUBSUB_MONITORING=ON -DUA_LOGLEVEL=100 -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 23.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tukotech, 2023-09-01:
I added the following to examples/pubsub/tutorial_pubsub_mqtt_subscribe.c and now I can see debug logs. Is this expected? 

```
    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    config->logger = UA_Log_Stdout_withLevel( (UA_LogLevel)100 );
```

neojaw, 2023-09-01:
I guess it is. The `UA_LOGLEVEL` compile option controls what log messages are generated at all, but the logging plugin (The Stdout logger is the default logging plugin) can still decide which ones to actually show / record etc. In the default server config, the logger is set as `UA_Log_Stdout_withLevel(UA_LOGLEVEL_INFO)`, so it will only print Info messages and upward.

However, I'm a bit confused why your use of `100` works here, I think you are supposed to use the actual enum constants (`UA_LOGLEVEL_*`) which are, as integers, way smaller than 100. You can take a look at `examples/server_loglevel.c` for an example.

tukotech, 2023-09-02:
RE: You can take a look at examples/server_loglevel.c for an example
Yes, that's where I learned about UA_Log_Stdout_withLevel. The level was coming from command line argument. 

https://github.com/open62541/open62541/blob/3ad448ec722087a369c8b2c94391dfe944679ed4/examples/server_loglevel.c#L38C1-L38C63

o6-nbeier, 2026-10-20:
Thank you for the question.

UA_LOGLEVEL only controls what is compiled in, while the default stdout logger additionally filters below INFO at runtime, so lower the logger’s threshold.

Closing for now.

closed 2026-10-20 by o6-nbeier